slic3r/Slic3r #3082: Infill overextrusion

legarasta, 2015-10-04:
Hello there!

With slic3r 1.2.9 i'm having constant over extrusion of infill when the area is 100% filled. This causes material to pile up and later nozzle shake by bumping into previous layer.

I believe i'm not overextruding in general since: i get perfect top surfaces and walls and this only happens for 100% filled zones.

Some photos to illustrate, my settings as well as the gcode i'm actually printing (white part).
[image]

Another part..
[image]
In this case, the code shows 3 lines on this wall:
[image]
..but the center one is deposited over the other two, resulting in that massive pileup..

Anyone getting this?

Tks in advance.

hoh61, 2015-10-05:
Hi,

I'm new within the 3D print community and I'm steadily learning the do's and donmt's and tricks necessary for a good printing. One thing I've learned very early is the impact of the extrusion multiplier. There is a variety in parameter setting found in the community. I experienced by myself the large variation of this parameter to get a good print result. Unfortunately the setting is strongly dependent on the used material (and probably on the storage condition too: e.g. moisture content). 

So for what I see on your picture you may improve your result by a small reduction of this parameter. Althought the manual says, that there shall be only small variations around 1 for this parameter, I needed for the original Vellemann PLA a setting of approx. 0.7 (still to be fine tuned). This value was reported in the community already for the usage with the cura engine too.

Personally I'm working on a procedure to establish the proper basic settings by a few test prints, because the large variety and the missing informations on proper settings.

legarasta, 2015-10-06:
I have already fine tuned it. It's not it; otherwise prints wouldn't be perfect in all other situations.

pyrophreek, 2015-10-06:
See issue #2960

hoh61, 2015-10-06:
Just for information: how did you solve it, which were the main changes?

pyrophreek, 2015-10-06:
at the moment, I am not using 1.2.9, as this is a deal breaker bug for me.  I have just been using 1.2.7 for now.  As it stands, this is a critical problem, and for a lot of people it makes 1.2.9 unusable.

legarasta, 2015-10-06:
@hoh61 this particular problem i discribe here i haven't been able to solve it. As for the extrusion multiplier i decreased it bit by bit until it was ok, and past that, and then got it back up until it came out perfect.

@pyrophreek Tks for the tip, as i suspected it's a bug..

Thank god for that, in a way! I built my "giant" custom printer from scratch and i couldn't figure out what the hell i was doing wrong. I'm gona proceed with test with 1.2.7 and si how it comes out.

Tks for your help!

mrvn, 2015-10-08:
Same problem here. I first thought I was extruding too much in general and fudged the extrusion factor. But that leaves gaps in other places and filament doesn't stick in concave places. What actually worked was fudging with the model making walls slightly thinner (it doesn't over in-fill) or thicker (there is enough space for the in-fill). But that is a lot of painful fiddling till the slic3r likes the model.

curiouspl2, 2016-01-03:
+1 for this.  it's pretty showstopper for almost every newbie.

airiclenz, 2016-01-26:
+1 
I am having the same problem. The wall thickness is exactly trimmed to the nozzle size (e.g. 3 shells on both sides plus one filling line) and it still does over-extrude. I'll try 1.2.7 as well. Is the bug present in 1.2.8 as well?

lordofhyphens, 2016-07-12:
95% sure this is the gapfill bug (which was fixed). Please try with 1.3.0-dev and let me know if it's still broken.
See https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-150# and please submit a screenshot from Slic3r if it's still a problem in 1.3.0-dev.

closed 2016-07-12 by lordofhyphens

bubnikv, 2016-07-12:
There is an issue with the rectilinear fill of narrow regions. The overfill
is caused by the connecting lines, which follow the perimeters. These
connecting lines are not accounted into the infill percentage and if the
region is very narrow, they may even overlap. The behavior may be
visualized by the extrusion simulator in the prusa3d fork, playground
branch. Regarding the gap fill, i don't consider it to be fixed. It will
now rather underfill.

<https://github.com/alexrj/Slic3r/issues/3082#issuecomment-231921489>, or mute
the thread