slic3r/Slic3r #3083: Feature request: Printer awareness

clefranc, 2015-10-06:
Hi @alexrj Did you have time to look at https://github.com/clefranc/Marlin4Delta/wiki/Printer-awareness

I've sent you an email with more details (or go to https://github.com/clefranc/Slic3r/issues/1).

Thanks
Christian

Mikk36, 2015-10-06:
what's wrong with just including those commands in your start and end gcodes?

clefranc, 2015-10-06:
I can include M2 & M3 (as explained in the wiki), but I wish M4 can be produce with proper value so I don't need to copy/paste into the generated G-code.

clefranc, 2015-10-06:
Another solution can be to include start/end script with some sort of place holder, like {totalextrusion}. I'll be able to add M4 command in slic3r start g-code:
```
M4 E{totalextrusion}
```

a4jp-com, 2015-10-06:
Is Alex still around?

clefranc, 2015-10-06:
Don't know, sent him an email a week ago...

alexrj, 2015-11-04:
@clefranc, Slic3r doesn't know the total E until it has generated the whole G-code.
Also, it's unclear what value would you expect in case of multiple extruders.