slic3r/Slic3r #3084: Feature Request: Don't support bridges shorter than <xyz> mm

mrvn, 2015-10-06:
If one has a model with a short bridge (say 5mm) and a long bridge (say 50mm) then one has to use supports. But that also adds supports for the 5mm bridge, which is quite pointless. Even harmfull given how hard it is to remove such a support.

The "Don't support bridges" option should ask for a length.

lordofhyphens, 2015-10-09:
From my perusal of SupportMaterial.pm, it looks like it removes all bridges from consideration when "dont support bridges" is selected. I am going to look around and see if we can detect the distance between anchors for this purpose.

burdickjp, 2015-11-02:
I was thinking about something similar while doing some overhang testing. My idea is to have a two column table of values. One column is the angle and the other is the length to allow without support. When graphed, this would make a curve. Anything above that curve would be supported, anything below that curve would not. This would allow long runs with nearly vertical angles and short runs with nearly horizontal angles.
There'd need to be some clever coding for things like arches, where you'd have changes in angles, but it'd be easy to visualize. You'd plot on our path and it'd show where you'd start to need support, at least with respect to the constant-angle testing you did.

jfpion, 2015-11-26:
I supporte this idea it is painfull to have to clean a part from all the little useless bridge only to have the support you need to support the big bridge. I don't think you really need to support bridge smaller than 5 mm for exemple, even in worst case printer (or you need to seriously re think your configuration)