slic3r/Slic3r #3085: Feature Request: more config options for supports

mrvn, 2015-10-06:
There should be an option to give the minimum distance between the support and object (in x/y plane). Also there should be an option for how much larger than the supported thing the support should be. I find the small size realy hard to grip and wouldn't mind a few extra mm.

Lord2Vader, 2015-10-14:
Yeah right. Wanted to support spiked hanging down (fingergrip handle) and the two small lines generated at the bottom was to small to hold cause the support getting 8 times higher than the width of the first layer. So the printhead knocks over the support causing the print to fail (and layershift)...

digiexchris, 2015-10-20:
and also allow printing support on top of a raft or brim rather than the raft or brim going around the support. I use a 1 layer raft due to a poorly built delta not being able to get good adhesion in some areas but being fine in others, so if I could print the support onto the raft it would be far more stable.

mrvn, 2015-10-22:
Although this is a bit of the opposite of the original feature:

Against the printer knocking over support it could help for the support to be connected to the object every X layers. Even just a single strand of filament connecting the support and object would add a ton of stability. More so if the strand bridges a gap with the support as column in the middle (connects to the object on 2 sides, even better 3, 4 sides), If the support uses pillars then interconnecting the pillars every X layers would stabilize them too. Print them as a H instead of I I.

VanessaE, 2016-01-18:
:+1: to all of mrvn's and digiexchris' ideas.  These things are needed pretty badly; as I write this, I'm printing a horse, which has a number of very narrow support columns (rectilinear mode), a few of which literally fell over from the vibrations of the table (not knocked over by the extruder) during the print because they didn't stick to the bed well enough, compared to the wider parts.