slic3r/Slic3r #3086: Feature Request: speed setting for concave perimeters / large direction changes

mrvn, 2015-10-06:
I'm printing a gear made by aternatively adding and subtracting cylinders around a larger cylinder to get rounded teeth. When printing the concave parts of the outer perimeter the tention in the extruded filament easily pulls the thread straight cutting across the teeth gap. A slower printing speed for concave parts should give the filament a better chance to stick and cool down in the right place.

Similar when there is a large direction change, e.g. 90-180° change in direction, the corner becomes round. Again a slower speed should give better quality there. But maybe this is already covered by some combination of the advanced acceleration settings?

phizz166, 2015-10-07:
I believe the inherent problem here isn't anything about the speed, but that the filament isn't sticking to the previous layer. Maybe it's too thin to get good interlayer adhesion. Play with your extrusion multiplier?

In general, you shouldn't need to slow your print down to get square corners if you're printing at any reasonable speed. (If you're printing at 200mm/s or something, well, I can't help you there)

mrvn, 2015-10-07:
It does have something to do with thinness. It works with thicker layers but gets problematic trying to print 0.1mm. On the other hand on other places (when it builds a solid wall, in the gui 2 perimeters, a blue line and 2 perimeters again) I think too much material is extruded leaving a bumpy surface/outside wall. So I don't think the extrusion multiplier is the right answere. But I can try. I will play around with the options some more, esspecially acceleration. Haven't touched them yet.

mrvn, 2015-10-11:
Seems I just hit the bug with overextrusion on small walls that was throwing me off more than I thought. I tried an older slic3r with more extrusion and it worked out mucht better.

closed 2015-10-11 by mrvn