slic3r/Slic3r #3087: Gap in the perimiter

veneridesign, 2015-10-08:
The slicer not make continuous perimeters but make gaps and fragmentations.

[image]
[image]

Mikk36, 2015-10-08:
STL and gcode files?

gege2b, 2015-10-08:
Also, you should not rely on an external toolpath viewer, but on slicer's one (only him knows stuff related to extrusion width etc...)

mrvn, 2015-10-08:
If only it could show tool paths, maybe even animate the path (as in draw is slowly so you see where it is going).

gege2b, 2015-10-08:
I don't get it, Slic3r doesn't show you the preview or layer tab ?

Have you actually printed an object showing this "holes" ?

veneridesign, 2015-10-08:
I'm using Slic3r for SLA/SLS and if the perimeter is segmented, even the laser stop and start...(bad quality surfaces..)
The real point is why other slicers make continuos lines and here the perimeter it s so messy?

Mikk36, 2015-10-08:
Could you supply the STL and gcode files where you have this issue?

Dongzhixiao, 2015-10-08:
I have met this problem before. It looks like showing the Gcode format in the Repetier-Host software. Maybe the increment of E between two points of the little arc is too small to show the figure.

Trellian, 2015-10-08:
This effect can also be caused if you are using the 'Pressure Advance' option.  It will cause the extrusion to not appear properly in the preview, although it will print correctly (IF you have set the P-A value correctly, that is)

veneridesign, 2015-10-08:
Here the files:
https://www.dropbox.com/s/3b6ok7zs81ow1je/6_Watch_Stand.STL?dl=0
https://www.dropbox.com/s/o13kayjj0q8996v/6_Watch_Stand.gcode?dl=0

Not sure.., thare are increments much smaller but perfecly shows, the strange things it's that these difects not start at the beginnings but only after some layers, curve withe the same raduis at the beginning are ok but going on became messy, not pressure Advance option used. I noticied that thi effect it's related in some way at the perimeter width parameter..

a4jp-com, 2015-10-08:
File > Export Config... ;)

veneridesign, 2015-10-08:
Ops...
https://www.dropbox.com/s/wm4ohsdpk68x6oh/config.ini?dl=0

Mikk36, 2015-10-08:
All of the config is included in the gcode file always though.

alexrj, 2015-11-04:
@veneridesignf, 404 not found