slic3r/Slic3r #3089: Feature Request: Please move "Maximum Volumetric Speed" to "Printer Settings"

jeder73, 2015-10-10:
I am using two K8200 printers, one with the standard nozzle (0.5 mm, drilled to 0.6) and one with a E3D V4 Hotend (0.4 mm). The extruder (gears similar to Wade's) are the same, just different stepper Drivers, A4988 with the bigger nozzle (1.8 A) and DRV8825 with the smaller (2.5A).
I noticed, that I need less maximum volumetric speed with the smaller nozzle (That's why I use the stronger DRV8825, of course with the higher frequency mod becaus of the low Ohm of the Steppers, but not strong enough). I think, it is a normal behaviour.
And because usually the extruder nozzle is mounted in a printer and it is possible to mount two different nozzles with two extruders, IMHO it would be better to move the "Maximum Volumetric Speed"-Setting (also the "Max Print Speed"-Seting) to the specific extruder settings of the specific Printer Settings.
Then it would be possible to adjust a model-dependend "Print Settings" profile, which is nearly independend of the printer and adjust printer-dependend settings in the "Printer Settings"-Profile

(I hope this not a duplicate feature request, I searched all Issues with "volumetric" in title and Body)

jfpion, 2015-11-26:
the pb is the same than the retraction option, it depends of the nozzle and of the filament , so there is no ideal place to move it to.
may be an average limit in the printer setting to choose the worst case filament, and a bonus option in the filament one.
il seems obvious that a pla filament would melt easily and the polycarbonate will need more heat to melt, and the thermal transfert will be more diificult to achieve from 20° to 290 than from 20° to 170 so for the same print speed you will need to heat more AND to have a longer thermal transfert zone, so there is an adaptation to do dependent of the filament.

i doubt there ise an easy way to model the diminution of the max volumetric speed function of the extrusion temperature .

lordofhyphens, 2016-07-11:
For printer settings, you can usually limit the max volumetric rate if you're using volumetric E on the printer firmware with M203 V<amount>

I can agree though that it could be added to Filament Settings and using the lowest of the values available. That way you can override for individual filaments but can have a default max for a given layer height. That or move it from Print Settings over to Printer/Extruder and have the override in Filament Settings.

Any thoughts about this @alexrj

VanessaE, 2016-07-11:
At that point, wouldn't it make more sense to just put the setting only with the Filament profile(s) and just propagate either a default value (as stored with the "- default -" profile) or the last one used, when creating a new profile?

lordofhyphens, 2016-07-11:
@VanessaE all profile options have defaults :)

My suggestion stems from the fact that it has some business living with
both printer settings (although that can be set with smoothieware via M203
if you're using volumetric extrusion) and filament settings as it tends to
vary by material as well as nozzle diameter. So there is (at least to me)
some call for having a general setting in Printer and a specific override
setting in Filament, the idea being that the Printer max speed is the
fastest you trust it with any filament.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

VanessaE, 2016-07-11:
Of course, your theory makes sense.  I was just thinking from the standpoint of "why have it in two places".

Situations like this (and my speed-affects-temperature suggestion) kinda make me wish Slic3r had a "dump it all into one section" option, with variables and settings to allow some things to vary based on layer height, print speed, nozzle diameter, or whatever.

dwillmore, 2016-07-12:
My workflow may not be representative of all people, but I think it's common enough to consider.  My printer can print 9mm^3 of PLA or ABS, but only 7mm^3 of PETG--and those values vary a little within a filament type.  It would help my workflow if the volumetric limits were at least overridable by a per-filament setting.

Thanks for the great sicer.

expertmm, 2017-12-01:
There is no way to implement this without making assumptions about filament. What varies little today may vary a lot years from now (such as if a new filament comes out that is common enough). If you are making a custom profile you should use the volumetric speed setting in the filament settings. The workflow of slic3r is to select a printer and select a filament.

OrthogonalConcept, 2018-07-30:
Add extruder-specific max volumetric speed override please.

I am using two different nozzle sizes in a dual extruder setup too. I am extruding fine features with TPU in the smaller, 0.3mm nozzle, and a very large piece of  PETG with a 0.5mm nozzle on the other side. The TPU has to be printed slowly, but the PETG can print 20 times faster. Since my max v speed override of 0.2mm^3/s for the TPU takes over the global setting, my part takes more than 100 hours to print. When I turn off the override, the part prints in 8 hours, but the feed is too fast for the TPU and I get crazy-lots-o-gaps in the fine features.