slic3r/Slic3r #3092: Feature Request: Insert Pause and/or custom Gcode between layers in 3D preview

pyrophreek, 2015-10-12:
Often it is useful to pause the print to change filaments, or to insert components into the print.  While this can be done by many different methods, such as manually editing the Gcode, or using a script, it would be nice to have a simple way to do this in Slic3r.

I propose adding a "insert pause script between layers"  button to the 3D and 2D preview windows. 

Functionality would be:

1) move layer slider to the desired last layer before the pause script
2) click the button
3) Dialog box opens with default pause script, which can then be edited if desired, and can also be saved as a new default script
4) click accept (cancel also available, as well as a button to save the current code as the new default)
5) a mark will appear on the slider at the current height indicating the presence of a pause script, which if clicked on will open the pause script to allow editing or removing of it.

First mentioned in issue #1055.

Better implementation possible if implemented with or after #2979

mrvn, 2015-10-16:
I think this would be better on a per object basis. If you print multiple objects one at a time then you might not want to pause at the same layer for every object, only for the one where you want to insert something. This could still use your interface idea (which I like) but also have an object selection.

pyrophreek, 2015-10-17:
@mrvn 
That is an interesting Idea, although arguably unnecessary, as you would still need to pause at each of the same heights overall (you only need one pause per height, whether it is for one object, or more).  Looking forward though, if future 3D model formats support tags for where pauses should be made to insert parts, this might be the better implementation to support that.

pyrophreek, 2015-10-17:
@mrvn 
just realized you meant for sequential printing >.< in that case having a per object setting would be beneficial.

Ianmcmill, 2016-09-01:
This would be a nice feature for inserting captive nuts like in this thing: 

1) proof of concept: http://www.thingiverse.com/thing:1633398

2) used in part: http://www.thingiverse.com/thing:798076   part: FCE.1501.01.005_fix_r01.STL

lordofhyphens, 2016-09-01:
Someone's welcome to try to work out the UI code for it (and it is 90-95% UI code work), and I'll review a pull request.

Darker, 2017-02-10:
It would be nice if - at least - slic3r allowed to you find particular moment and locate the G-code line for it.

goofdad, 2017-10-31:
[3092-pause.zip](https://github.com/alexrj/Slic3r/files/1432001/3092-pause.zip)
This was an itch I needed to scratch :)

Attached is the diff as I don't have access to create a branch ... but it's been a while since I worked with git. If there's a way to get this into a PR, let me know.

lordofhyphens, 2017-10-31:
@goofdad To create a Pull Request:
1. Fork Slic3r on Github. 
1. Switch your local checkout to point at that with [`git remote set-url`](https://help.github.com/articles/changing-a-remote-s-url/). 
1. Make a new branch (`git checkout -b 3092-pause`).
1. Ensure that this branch has your changes 
    * If you were working on `master` before and haven't done a commit yet, then the new branch will have your work in it.
    * Commit your changes to your feature branch.
1. Push your feature branch to your fork of Slic3r (`git push origin 3029-pause`).
1. Create a new pull Request on alexrj/Slic3r from your feature branch. https://help.github.com/articles/about-pull-requests/

goofdad, 2017-10-31:
Done.
https://github.com/alexrj/Slic3r/pull/4177

tcurdt, 2018-09-09:
Any progress on this? It's been a few years and this would be really handy.

lordofhyphens, 2018-09-09:
@tcurdt @goofdad made a pull request to implement, I requested changes and nothing else got finished. You are welcome to take a look at what he did and make the requested changes to finish it.

jwynn6, 2019-05-01:
Can we just get an interface exactly like the cut  interface? Just change the name to pause?