slic3r/Slic3r #3093: Slic3r for Windows crashes when modifying field values

dayssincethedoor, 2015-10-12:
I have noticed that while changing values for many fields Slic3r will crash if the value is even briefly 0. For instance today I was changing the bed size on my printer in Slic3r. I was changing it from 300x300 to 300x100. When I backspaced the 3 out of the y value leaving 00, before I could type the 1 Slic3r crashed. This happens with a number of different fields. To get around this I have to first change the y value to 1300 and then back space the 3 to get 100 as the final value.

This tells me that there is something attempting to make calculations after each key press in a field and is getting a division by 0 error. It should be doing it after someone is no longer editing the field. It should also properly check if a field is 0 and post an error rather than using 0 for the calculation.

I am in the process of looking through the code to see if this is something I can help fix, but I am very new to the project.

lordofhyphens, 2015-10-12:
That input validation is done in perl. I thought it had been fixed, which
version are you usiing?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

dayssincethedoor, 2015-10-12:
The 1.2.9 release

alexrj, 2017-03-18:
We have nice stable automated builds now, available at http://dl.slic3r.org/dev/.
Feel free to reopen should you notice this issue again.

closed 2017-03-18 by alexrj