slic3r/Slic3r #3095: Feature Request: distance between support and object

mrvn, 2015-10-16:
I'm printing a 40 teeth GT2 pulley with a hole for a M8 nut in the bottom (no, I can't print it upside down). The support being added is placed all around the inside of the hexagonal hole just a miniscule distance from the object. This causes the support to fuse with the object during print making it hard to impossible to later remove the support. It would be so much better if there were a setting for a minimum horizontal distance between the object and support.

KiteLab, 2015-10-16:
How about adding a one layer diaphragm on top of the nut, and then printing without support?
If that works well enough depends on your ability to print bridges.

pyrophreek, 2015-10-17:
Sounds like one of those things that is relatively part specific, so better to include the support in the model rather than rely on the slicer to generate it.  
Still, it is an interesting idea.

mrvn, 2015-10-19:
I don't think this is part specific at all. I had problems with the support being too near my object in basically every case I used support. Unless the support is inside a hole you can usually break it off but more horizontal clearance would be better. I'm fine with printing a 1-3mm bridge so I don't need support to be 0.1mm from the object.

lonelymyp, 2015-11-11:
Fuse with the object arise from problems with the mechanics. check belt tension, etc.
or do the jumper, similar like here: http://www.thingiverse.com/thing:135799
they are easy to break down after printing.

mrvn, 2015-11-14:
It might be the fault of bad mechanics but that's the quality I have. So it falls to the software to work around it. (Not sure what that extruder link was ment to be)

lonelymyp, 2015-11-14:
this is an old problem. but Alex does not want to solve it :(
https://github.com/alexrj/Slic3r/issues/2916

lonelymyp, 2015-11-14:
thin membrane then easily removed. model can be printed without the supports, or may be added to the support that is easy to break off.
[image]

mrvn, 2015-11-16:
I tried adding "thin" membranes. But "thin" is relative. If the membrane is to thin the slicer skips it altogether. If it is to thick it will have many layers and has to be cut with difficulty or drilled. You have to adjust the membrane to the layer height in the slicer. And even then I always end up with walls 2 lines thick or membranes 2 layers thick. Most of the time all that would be needed would be 2 or 3 strands for the rest to hold.

lordofhyphens, 2016-04-15:
Duplicates #2916

lordofhyphens, 2016-06-28:
Please use #1539 for further discussion.

closed 2016-06-28 by lordofhyphens