slic3r/Slic3r #3100: [feature request] fine tuning flow rates of individual features

curiouspl2, 2015-10-24:
Hello. I've encountered many times the situation where one would like to fine-tune just few features to make print perfect, and it's not very possible by tuning just extrusion width.

The problem originates from many reasons, for example extrusion width vs real extrusion width varies from nozzle to nozzle , also some border settings (like very high or very low) cause some under or over extrusions. 
It might be print specific or surface specific - for example sometimes one wants top solid fill to be very tight and even overextruded to form nice flat and watertight surface, and sometimes one wants it under-extruded _and_ with very fine extrusions to make nice visual pattern - often together with concentric fill. 
Currently one can tweak only extrusion rate globally which is often not enough. 

greetings

neumond, 2015-11-08:
I support this. Solid fill pushes too much plastic and I have overflows. With any flow width setting. Forced to fix gcode using scripts.

mrvn, 2015-11-17:
+1 too. With an extra thought thrown in. Is it really (just) the extrusion amount that is relevant here? Maybe a setting for overlapping is more relevant. (Too much) overextrusion will leave a bumpy surface so I don't want that. But by printing the top/bottom layer with slightly overlapping lines the bonding between the lines should increase and form a better seal.

lordofhyphens, 2017-10-24:
@mrvn overlap perimeter-perimeter, or perimeter-infill?