slic3r/Slic3r #3101: Changing the filament temperature for "other layers" causes an unintended tool change.

Lodran, 2015-10-24:
On a dual carriage, dual extruder printer, I've set the filament temperatures to "First layer: 205" and "Other layers: 220".

I've set "Tool change G-code" to G199, which is my own gcode for parking the current tool (and is carriage) out of the way of the other carriage.

The gcode generated at the first layer change is as follows:

G1 X104.146 Y129.919 E30.22393
M104 S220 T0 ; set temperature
G1 E29.72393 F900.00000
G92 E0
G1 Z0.500 F7500.000
G1 E-9.50000 F900.00000
G92 E0
G199
T0
G92 E0
G1 X96.279 Y130.679 F7500.000

At the beginning of this code, the current tool is T1, and T0 is parked offsides on the left hand side of the printer.

When the M104 line is reached, smoothieware makes a tool change (as specified by the NIST gcode standard).  That tool change was made without parking T1.

When the G199 command is reached, the printer parks T0 (which is already parked).

The T0 command has become a NOP, because the printer has already switched tools.

When the final G1 command executes, the T0 carriage crashes into the T1 carriage, because T1 was never parked.

If the M104 command is removed, this code works as expected - when the G199 command is reached, it parks T1 on the right hand side of the printer in preparation for the T0 command.

mdebski, 2015-11-04:
I can confirm the issue. ex. Smoothieware does not support "M104 Snnn Tn".

alexrj, 2015-12-11:
According to the official answer from Smoothie devs in https://github.com/Smoothieware/Smoothieware/issues/760, that syntax is supported in Smoothie. Can we close this issue?

mdebski, 2015-12-11:
No, the official answer [0] is "it's not supported and won't be"
They interprete M109 S123 T1 same as "T1 \n M109 S123" so it causes toolhead change, while slic3r assumes it doesn't.

[0] https://github.com/Smoothieware/Smoothieware/issues/760#issuecomment-154208693