slic3r/Slic3r #3103: Slicing issue

nenadalm, 2015-10-27:
Hi. I have issues with slicing one objects (details below).

**versions of used programs**
* [Slic3r](http://slic3r.org/): 1.2.9
* [OpenSCAD](http://www.openscad.org/): 2015.03-1 

**files**
* [scad file used to generate stl using openscad](https://mega.nz/#!oQwSTIBY!opYkUxWVfkJgDy29b5uzWiE2POERqbvUhUDlC7HD0Oc)
* [generated stl](https://mega.nz/#!9M4llDKC!PdBQI76pM4A5Sgcxe0oWbmNPausNJ5JXChVCzER7byE)
* [png showing how the object should look like (taken from openscad)](https://mega.nz/#!FRwmSYyL!jkpLHL6EB3SxwHFmxI7umU6iYAnc4Tkrw5ErlyFfy-o)
* [png showing how the object was sliced (taken from preview of slic3r)](https://mega.nz/#!EBg02KBK!zCfOoL6_0JtWZpGdxQv4LRHSdZjhMFQo5eM8HUwfxPs)

t-paul, 2015-10-27:
That's probably not a slic3r issue, the geometry generated by the platform() module seems to have very thin walls between the segments. That causes the model to come out non-manifold.

nenadalm, 2015-10-27:
thanks for your response, I found something about it and will try to check it: https://blender.stackexchange.com/questions/7910/what-is-non-manifold-geometry

nenadalm, 2015-10-27:
Thank you very much! Not sure why I had thin walls there (I thought that it's ok if they dissapear on rendering - I saw them just in preview).

I used better polygons instead of some strange polygon diffs I used before and now it works: https://mega.nz/#!dVw2xJSK!_OwHRpW8QOjCk9j75EP6g3AJfjA-Wns0bzcLmictUjs (or diff here: https://github.com/nenadalm/3d-print/commit/f3fbd4d45efa5c52feeb2b3ce8bedff8593771fe)

Now going to read the link I posted :)

closed 2015-10-27 by nenadalm

t-paul, 2015-10-27:
Another option is to generate the full polygons first using the new list comprehension features and extrude to 3D last, e.g.:
```
module platform(d) {
    r = d / 2;
    angle = 360 / chunks;

    linear_extrude(height = grille_fill_h) difference() {
		polygon([ for (i = [0 : chunks - 1]) cartesian_from_polar(length = r, angle = i * angle) ]);
		polygon([ for (i = [0 : chunks - 1]) cartesian_from_polar(length = r - thickness, angle = i * angle) ]);
	}
}
```