slic3r/Slic3r #3104: Huge "blops" at start of external perimeter at every layer

justmyopinion, 2015-10-29:
Slic3R 1.29 W7/64, Extruder direct drive, ABS/1.75 230deg at different speeds
This was something I tried to get rid of by tuning retraction in earlier versions, but this one seems to be very persistent and it is not possible to remove by simple changes (as far as I know). I tried with auto speed settings and without, same mess.
if i print a cylinder of 2 walls (outer perimeters) with fillin and select random start i get a huge blop every time outer perimeter starts and it cannot be removed by changing retraction. 
When inner perimeter has been printed, fillin is next printed between walls and after this extruder travels to the next point at outer perimeter and make a layer change and start printing new perimeter. This is where the BLOP is born.
It looks like extruder has a huge pressure built up during fillin and this is released during/after travel and land at the perimeter start despite retraction, at least that is how it looks to me. 
I do not know whether this came with 1.29 or earlier, but it is rather annoying

a4jp-com, 2015-10-29:
Have you tried the new developers version 1.2.10-dev?

justmyopinion, 2015-10-29:
No,  my dev environment is not updated for this.
Should it make a difference?

a4jp-com, 2015-10-30:
I'm not getting this problem but I use PLA.

a4jp-com, 2015-10-30:
Can you post your settings and a model you have problems with?

Paul--W, 2015-11-02:
This appears to be the same issue I investigated recently on Reprap.org, with photos, source files, and analysis.  See http://forums.reprap.org/read.php?262,550754

a4jp-com, 2015-11-02:
Sorry but that link doesn't help. Where are your configuration files for Slic3r?

lonelymyp, 2015-11-11:
Too little jerk causes such a problem