slic3r/Slic3r #3106: Feature request G-code flavour

justmyopinion, 2015-10-31:
In printersettings you may select your G-code flavour before exporting which is a fine feature.
It is also fine that you may select no extrusion, but maybe this should not be in this particular menu?
If i select no extrusion which could be fine for machine experiments i do not know which flavour the rest of G-code may have. I may assume that G-code flavour is not all about extruder coding?.
Maybe disabling extruding should be under extruder selections?

alexrj, 2015-11-04:
What's your use case?

justmyopinion, 2015-11-04:
Well it was of a "sort of" case.
i was asked on how to slice for a CNC rebuilt for 3D-printing, which i also did as a project some years ago.
It was  a typical case that you would select MACH3 flavour output and for some time you may not even have an extruder so a switch for suppressing code for this would be nice for testing as this would be a stand alone installation parallel to CNC XYZ machine. 
But the SLICR excludes selection as you may select MACH3 flavour OR no extruder but not both.

alexrj, 2017-03-05:
This really looks like a very particular case. You can easily do it with a post-processing script which removes all the Ex arguments from commands, so I'd rather not change Slic3r. I hope you understand :)

closed 2017-03-05 by alexrj

lordofhyphens, 2017-03-05:
We have "no extrusion" flavor already that removes E axis.

I also have a PR for something else (pumps/solenoid extruders) that could
be abused to omit extrusion axis.