slic3r/Slic3r #3108: Bridge changes direction in midair

kroylar, 2015-10-31:
I'm trying to print the [Ultimate Extruder Calibration Test](http://www.thingiverse.com/thing:9804). Every bridge is like the one shown below. It begins from the anchor marked 1, changes direction at the point marked as 2, and ends at the point marked as 3.

[image]

I'm using 1.2.9 on Windows 7.

koghe, 2015-11-14:
Hi 
i'm using 1.2.10-dev i think i have the same problem 
[image]

stl https://www.dropbox.com/s/75pmpcwuaptfkuy/coolingFan.stl?dl=0
config https://www.dropbox.com/s/mza8hbtuhwjoh2y/config.ini?dl=0

a4jp-com, 2015-11-14:
I've just sent my printer away to get the z axis rod replaced. After that, I can check if I get the same problems as you if you like.

xoan, 2015-11-14:
I think it is the correct behavior to do this bridge.

Note that you are printing the inner perimeter first, so the toolpath for
the big hole is the one you show in the picture.

--
http://about.me/xoan

koghe, 2015-11-14:
@xoan sorry i don't understand, if i print with external first i get the right bridge?
lessbetter but is always present
[image]

kroylar, 2015-11-14:
@a4jp-com This does not require you to have a 3d printer. You can check for this problem by slicing a model and viewing the generated gcode.

a4jp-com, 2015-11-15:
How thick is the model in that green area? I'll look when I get home.

koghe, 2015-11-15:
@a4jp-com thank's for looking at it, the model is a openscad straight fan, in the 2nd image th same but with a involute fan
the green is the support auto generated by slic3r, hole diameter is ~20mm, fan is 70mm 
With straight blade the problem is more evident, less with involute  blade.

[image]
[image]

a4jp-com, 2015-11-15:
Screenshots of the file posted.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

koghe, 2015-11-15:
Thanks for reply! i cleaned the geometry and mesh density maybe is a modeling problem.

I tried your settings, when preview the gcode in gcode.ws it give a really strange result
in the time and layer count fields... have you tried?

[image]

don't know what happen here
i have to remove autospeed and now preview field show the time
and also Layer field show 95 of 96

[image]

also infill 30% scale is completely different from your conf...
[image]

after a complete new install of Slic3r and new configuration
linux ubuntu 15.04 
perl -v This is perl 5, version 20, subversion 2 (v5.20.2)

ini : https://www.dropbox.com/s/zeu3vmjcnhtqixz/config-autospeed-bug.ini?dl=0
stl : https://www.dropbox.com/s/eb7a36wj5c33p9g/coolingFan-new.stl?dl=0
gcode : https://www.dropbox.com/s/ut30b0o0n4tvznh/coolingFan-new.gcode?dl=0

maybe have to fill another issue, sorry to post here

a4jp-com, 2015-11-16:
[image]

This is what I saw on the gcode.ws site. If they don't have the time written maybe they have a bug.

perl -v

This is perl 5, version 22, subversion 0 (v5.22.0) built for MSWin32-x86-multi-thread-64int

Would you like me to try the second file you have?

I'm using Slic3r 1.2.10-dev.

koghe, 2015-11-16:
You have the same result like me, the second file is here , i think you mean this one?
this one is generated by lsetting 0 for all speeds
ini : https://www.dropbox.com/s/zeu3vmjcnhtqixz/config-autospeed-bug.ini?dl=0
stl : https://www.dropbox.com/s/eb7a36wj5c33p9g/coolingFan-new.stl?dl=0
gcode : https://www.dropbox.com/s/ut30b0o0n4tvznh/coolingFan-new.gcode?dl=0

gcode.ws is right, using the autospeed give me the result above 
If i try to print with this setting is awfull, first layer speed is normal, from the second layer 

this gcode look right, the difference is just the speed settings and not keep 0 on the fields.
https://www.dropbox.com/s/2qsne56oqofdpmz/coolingFan-noautospeed.gcode?dl=0

this is a fresh new slic3r clone

clough42, 2016-01-01:
Don't know if this is related, but I'm seeing similar situations in 1.2.9, where extrusions change direction in midair.  Something seems to be wrong with bridging detection.  I'm attaching these examples to a few open issues that look related.

[image]
[image]
[image]

Config and STL files:
[ShroudBridges2.zip](https://github.com/alexrj/Slic3r/files/75856/ShroudBridges2.zip)