slic3r/Slic3r #3110: 1.2.10-dev (Windows 10) - Infill patterns not working

a4jp-com, 2015-11-02:
I just updated to 1.2.10-dev (Windows 10) and the infill patterns not changing.

http://a4jp.com/3d-files/bugs/CUBE.stl

http://a4jp.com/3d-files/bugs/2015-11-02nd-05-52-pm-for-github-config.ini

a4jp-com, 2015-11-02:
Did the infill work with 1.2.9? I didn't try changing it so I'm not sure.

markwal, 2015-11-02:
It did, but perhaps you're below the 70 mm^2 threshold so it is always solid?  10x10 would be 100 minus the area covered by the perimeters.  What happens on a 20x20 cube or if you set the solid infill threshold to 0?

a4jp-com, 2015-11-03:
I did a 20x20mm cube and it worked. I hope the smaller infill can get fixed soon. I make little houses and the diagonal fill is really bad on these little houses. I need a way to select other fill patterns if possible.

markwal, 2015-11-03:
You can just set that solid infill threshold to something lower like 30 it's right there in your picture

https://github.com/alexrj/Slic3r/issues/3110#issuecomment-153305360

a4jp-com, 2015-11-04:
I set the threshold to 0 and got the infill settings back but I just want to know why it is set to 70 by default. What problem is this trying to fix? If the program disables the infill pattern could it gray out the box and say "decrease the solid threshold area to allow fill patterns"?

markwal, 2015-11-04:
It doesn't know it is disabling the infill for the whole part. It only does this per each infill area per layer. And your settings aren't for only one part anyway.

This is to prevent weirdness when the infill pattern, low infill and a very small area combine to make the infill absurd. For a bot that isn't under or over extruding, solid infill helps improve quality without to much expense because the area is small.

https://github.com/alexrj/Slic3r/issues/3110#issuecomment-153649143

a4jp-com, 2015-11-04:
@alexrj what do you think about the default number 70? Is that best?

alexrj, 2015-11-04:
It has proven to be reasonable for several years.

closed 2015-11-04 by alexrj