slic3r/Slic3r #3115: Feature Request: "support" infill pattern

mrvn, 2015-11-05:
Sometimes I would like to print without any (or very little) infill. Problem then is that nearly everything but a totally flat top will not work at all. E.g. a small cube on top of a large cube will just fall down.

What I would like to happen is to add support inside the object to support that small cube and help the top layers of the large cube to form properly. Basically extend the perimeter of the small cube down through the big cube till it hits bottom. Since the support would not mend to be removed and wouldn't be removable anyway (it's inside the object) the Z separation makes no sense there. We want the inside support to layer bond. Apart from that maybe the pillar support type would already work as support infill.

lonelymyp, 2015-11-11:
http://slic3r.org/blog/modifier-meshes

mrvn, 2015-11-14:
How does that help? Are suggesting to select 0% infill for the object and then manually add meshes for all the parts where there should be support? Then I might as well design the whole thing hollow with structure inside in the first place.

lonelymyp, 2015-11-14:
Exactly. 0-5% for all items and 15-20% support where You needed.
You can add simple cubes and cylinders. Making them take just a minute.
If you going to draw the structure, then they will be monolithic, and you have to design a hollow model, shell, it is difficult.

mrvn, 2015-11-16:
That won't work at all, or nearly never. When you put a small cube on a large cube you need to support exactly on the border of the small cube. With 15-20% infill you get maybe one or two lines or the whole cube fits in the middle of a hexagon and you get no support at all. In short: the infill does not adapt to the shape it needs to support.

ghuser2, 2015-11-17:
So the modifier approach would be to do the infill with several modifiers (layered), so that the higher you go the more infill there is to support the roof?

pyrophreek, 2015-11-17:
This is already implemented, just select "Only infill where needed."  Also, adding a few extra top layers helps.  In reality, for your example the best option is actually just to increase the top layers, and then just bridge the top.  If you have a more complex shape, then using infill with the aforementioned setting will also work.

lordofhyphens, 2017-02-08:
@mrvn have you given "Only Infill where needed" or increasing # of top layers a try?