slic3r/Slic3r #3117: WARNING: Boost libraries

alexrj, 2015-11-06:
This is a heads-up for the most active users. Sorry for abusing the issue tracker.

I will soon remove the Boost files from this repository for a few good reasons.
You will need to install the Boost libraries separately in order to compile Slic3r. It's not too hard.

Linux users can use their package manager. Debian/Ubuntu users will need to install the `libboost-system-dev` and `libboost-thread-dev`.

MacOS X users can use homebrew (`brew install --without-python boost `) or MacPorts (`sudo port install boost`).

Windows users can download the boost source package from here, then extract it into C:\dev\boost_1_59_0 (or whatever the downloaded version is), then open the prompt in that directory and type:

```
bootstrap.bat mingw
b2 toolset=gcc system thread
```

The Build.PL script should be smart enough to locate Boost in all the default directories. If you installed it to a non-standard directory you can set the BOOST_DIR environment variable.

Please report here any issue or hint so that we can make this transition as smooth as we can.

lordofhyphens, 2015-11-06:
Good to know, I'll make sure to update my package builder to add it as a
build requirement (if it hasn't got it already).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2015-11-09:
It looks like the Debian packages already had Boost
(specifically libboost-geometry-utils-perl) as a run-time dependency and
all headers as a build dependency.

1.2.10-dev package source is up on my fork.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-11-09:
libboost-geometry-utils-perl hasn't been needed for a long time, actually; it should be removed.

Stevie-O, 2015-11-21:
I followed the directions here: [https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows], using the Strawberry Perl option, and I'm getting errors relating to Boost.

When I built Boost, it gave me two directories: one for the compiler (include files), and one for the linker (library files).

If I set BOOST_DIR to the include file directory, the build fails right away with "Configuring C:/Users/Stevie-O/projects/external/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed."

If I set BOOST_DIR to the library file directory, the build fails with: "src\libslic3r\Point.hpp:127:37: fatal error: boost/polygon/polygon.hpp: No such file or directory"

alexrj, 2015-11-23:
@Stevie-O, where is your boost directory located and how is it named?

Stevie-O, 2015-11-26:
@alexrj,

C:\Users\Stevie-O\dev\boost_1_59_0

(so, the following file exists: C:\Users\Stevie-O\dev\boost_1_59_0\b2.exe)

`b2 toolset=gcc` gave me the following:

>The Boost C++ Libraries were successfully built!
>
>The following directory should be added to compiler include paths:
>
>    C:\Users\Stevie-O\dev\boost_1_59_0
>
>The following directory should be added to linker library paths:
>
>    C:\Users\Stevie-O\dev\boost_1_59_0\stage\lib
>

alexrj, 2015-11-26:
As advised in https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows you should put your Boost libraries in C:\dev

pzl, 2016-01-08:
Boost 1.59 has been compiling fine for me for a while.

Boost 1.60 on the other hand:

```
cc ... -o src/libslic3r/PolylineCollection.o src/libslic3r/PolylineCollection.cpp
In file included from src/libslic3r/Line.hpp:5:0,
                 from src/libslic3r/Polyline.hpp:5,
                 from src/libslic3r/PolylineCollection.hpp:5,
                 from src/libslic3r/PolylineCollection.cpp:1:
src/libslic3r/Point.hpp:123:12: error: redefinition of ‘struct boost::polygon::coordinate_traits<long int, true>’
     struct coordinate_traits<coord_t> {
            ^
In file included from /usr/include/boost/polygon/polygon.hpp:12:0,
                 from src/libslic3r/Point.hpp:117,
                 from src/libslic3r/Line.hpp:5,
                 from src/libslic3r/Polyline.hpp:5,
                 from src/libslic3r/PolylineCollection.hpp:5,
                 from src/libslic3r/PolylineCollection.cpp:1:
/usr/include/boost/polygon/isotropy.hpp:201:10: error: previous definition of ‘struct boost::polygon::coordinate_traits<long int, true>’
   struct coordinate_traits<long, sizeof(long) == sizeof(polygon_long_long_type)> {
          ^
error building src/libslic3r/PolylineCollection.o from 'src/libslic3r/PolylineCollection.cpp' at /usr/share/perl5/core_perl/ExtUtils/CBuilder/Base.pm line 173.
```

a4jp-com, 2016-01-09:
I'm still on version 1.59 without problems. Thanks for pointing this out @pzl. I was just about to update.

pzl, 2016-02-23:
deleting the following lets everything build fine on Boost 1.60 (all tests pass, nothing out of the ordinary when running)

```diff
diff --git a/xs/src/libslic3r/Point.hpp b/xs/src/libslic3r/Point.hpp
index af0f59a..d4eaa5f 100644
--- a/xs/src/libslic3r/Point.hpp
+++ b/xs/src/libslic3r/Point.hpp
@@ -120,16 +120,6 @@ namespace boost { namespace polygon {
     struct geometry_concept<coord_t> { typedef coordinate_concept type; };
     
     template <>
-    struct coordinate_traits<coord_t> {
-        typedef coord_t coordinate_type;
-        typedef long double area_type;
-        typedef long long manhattan_area_type;
-        typedef unsigned long long unsigned_area_type;
-        typedef long long coordinate_difference;
-        typedef long double coordinate_distance;
-    };
-
-    template <>
     struct geometry_concept<Point> { typedef point_concept type; };
    
     template <>
```

Not sure if it's the correct thing to do, but it's functional for the moment

forkineye, 2016-02-29:
Thanks @pzl, I noticed this as well with Boot 1.60.  Building now with the redefinition within Point.hpp commented out.

alexrj, 2016-03-13:
Thank you for reporting; this is the Boost commit that broke compilation with 1.60:

https://github.com/boostorg/polygon/commit/0ac7230dd1f8f34cb12b86c8bb121ae86d3d9b97

alexrj, 2016-03-13:
Okay, now Slic3r is compatible also with Boost 1.60.

closed 2016-03-13 by alexrj

llluis, 2016-03-21:
Just to add comment for other users: if you are on Windows and following the default instructions, *don't* set BOOST_DIR. 
Just leave the files at the default location (c:\dev\boost* or c:\boost*) and the script will find them (and remember to execute the boostrap).

Note to @alexrj : on Windows, if the user sets BOOST_DIR, the Build.PL will not work as it will add BOOST_DIR to @boost_libs, instead of "BOOST_DIR/stage/lib".

Tinchus2009, 2016-04-23:
Ok.. Im on win 10. I installed boost 1.60 (latest version to april 22 2016). It compiled ok, perl depencies are ok. Evrything is ok tip I coomand Perl Buil.pl. I get a final message saying boost libraries could not be found. Boost is installed in C:/dev?boost*. I tried: perl Build.PL BOOST_DIR=C:\dev\boost_1_60_0\ and perl Build.PL BOOST_DIR=C:\dev\boost_1_60_0\stage\lib
and I still get the message:

Configuring C:/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_DIR environment variable:

    BOOST_DIR=/path/to/boost perl Build.PL

N/A
! Configure failed for ./xs. See C:\Users\Martin\.cpanm\work\1461371932.8632\build.log for details.
The XS/C++ code failed to compile, aborting

Any ideas?

markwal, 2016-04-23:
You did this part, right?

bootstrap.bat mingw
b2 toolset=gcc system thread

Tinchus2009, 2016-04-23:
Yes I did. Boot compiled with no problems and it  is installes under C:\dev\...

bubnikv, 2016-04-23:
Try the Boost version 1.59.
Vojtech

bubnikv, 2016-04-23:
Also you may change the path to Boost by editing the file xs/Build.PL
For myself, I changed the following section to pull boost from d:\src

    if ($^O eq 'MSWin32') {
        for my $path (glob('C:\dev\boost*'), glob ('C:\boost*'), glob
('d:\src\boost*')) {
            push @boost_include, $path;
            push @boost_libs, $path . "/stage/lib";
        }
    }

Tinchus2009, 2016-04-23:
Ok. I will try 1.59. It doesnt work on 1.60. I even put the exact path on the xs/Build.PL: 

if ($^O eq 'MSWin32') {
        for my $path (glob('C:\dev\boost_1_60_0'), glob ('C:\boost*')) {

Boost IS there.... the /stage/lib is there too. ...

Tinchus2009, 2016-04-23:
Ok... Im doing something wrog and I dont know what. I compiled boost 1.59, I got this:
The Boost C++ Libraries were successfully built!

The following directory should be added to compiler include paths:

    C:\dev\boost_1_59_0

The following directory should be added to linker library paths:

    C:\dev\boost_1_59_0\stage\lib

I installed strawberry, then I went to c:\slic3r, did a Perl Build.PL and againg I got: 
Configuring C:/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_DIR environment variable:

    BOOST_DIR=/path/to/boost perl Build.PL

I follwed the instructions of: https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

Any ideas what Im doing wrong?

Tinchus2009, 2016-04-23:
Looks like it works on 1.59. Im stuck now with the Alien-wxgidgets.... 10 minutes so far ...:

c:\Slic3r>cpanm -f Wx
--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
==> Found dependencies: Alien::wxWidgets
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz ... OK
Configuring Alien-wxWidgets-0.67 ... OK
Building and testing Alien-wxWidgets-0.67 ...

vlast3k, 2016-06-10:
Hi guys,

Any hint what am i missing to built boost on windows 7 x64?
i am able to build slic3r up to the point where it requires boost
i downloaded the source package of boost

```
C:\dev\boost_1_59_0>bootstrap mingw
Building Boost.Build engine

Bootstrapping is done. To build, run:

    .\b2

To adjust configuration, edit 'project-config.jam'.
Further information:

    - Command line help:
    .\b2 --help

    - Getting started guide:
    http://boost.org/more/getting_started/windows.html

    - Boost.Build documentation:
    http://www.boost.org/build/doc/html/index.html

C:\dev\boost_1_59_0>b2 toolset=gcc
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:1072: in configure-really
*** argument error
* rule generate-setup-cmd ( version : command : parent : options * : cpu : global-setup : default-global-setup-options : default-setup )
* called with: ( 9.0 : C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC\bin : C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC : <command>C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC\bin\cl.exe : i386 :  : x86 : vcvars32.bat )
* missing argument global-setup
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:798:see definition of rule 'generate-setup-cmd' being called
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:201: in configure
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:153: in msvc.init
C:/dev/boost_1_59_0/tools/build/src/build\toolset.jam:43: in toolset.using
C:/dev/boost_1_59_0/tools/build/src/build\project.jam:1052: in using
project-config.jam:3: in modules.load
C:/dev/boost_1_59_0/tools/build/src\build-system.jam:249: in load-config
C:/dev/boost_1_59_0/tools/build/src\build-system.jam:412: in load-configuration-files
C:/dev/boost_1_59_0/tools/build/src\build-system.jam:524: in load
C:\dev\boost_1_59_0\tools\build\src/kernel\modules.jam:295: in import
C:\dev\boost_1_59_0\tools\build\src/kernel/bootstrap.jam:139: in boost-build
C:\dev\boost_1_59_0\boost-build.jam:17: in module scope

C:\dev\boost_1_59_0>b2 toolset=gcc
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:1072: in configure-really
*** argument error
* rule generate-setup-cmd ( version : command : parent : options * : cpu : global-setup : default-global-setup-options : default-setup )
* called with: ( 9.0 : C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC\bin : C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC : <command>C:\Program Files (x86)\Microsoft Visual Studio 9.0\VC\bin\cl.exe : i386 :  : x86 : vcvars32.bat )
* missing argument global-setup
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:798:see definition of rule 'generate-setup-cmd' being called
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:201: in configure
C:/dev/boost_1_59_0/tools/build/src/tools\msvc.jam:153: in msvc.init
C:/dev/boost_1_59_0/tools/build/src/build\toolset.jam:43: in toolset.using
C:/dev/boost_1_59_0/tools/build/src/build\project.jam:1052: in using
project-config.jam:3: in modules.load
C:/dev/boost_1_59_0/tools/build/src\build-system.jam:249: in load-config
C:/dev/boost_1_59_0/tools/build/src\build-system.jam:412: in load-configuration-files
C:/dev/boost_1_59_0/tools/build/src\build-system.jam:524: in load
C:\dev\boost_1_59_0\tools\build\src/kernel\modules.jam:295: in import
```

tareko, 2016-07-21:
I'm not sure which packages are actually required, but for me this didn't work until I added the `libboost-all-dev` package

tarek : )