slic3r/Slic3r #3118: flow math: improved model

hoh61, 2015-11-06:
Upon my first poor experiences with my new 3D printer I was keen on improving my results. My analysis showed me one topic, which I have analysed a little bit deeper: the applied flow math.
Within the documentation of slic3r there is one chapter dealing with this. For extruding on a flat surface it starts:
"In this case the problem is: what shape will our extrusion get? We know it will get squashed horizontally, but will it have a rectangular or oval shape? What's the maximum extrusion width we can get with a given nozzle diameter before plastic starts curling at the sides? "
The derived model is valid as long as the extrusion width is larger or equal nozzle width + layer height.
[image]
If the extrusion width is reduced below this critical value, the model assumes gaps below the nozzle width, which will not be filled with material. From physical point of view this cannot be explained.
[image]
So if we drive this model to an extrema, an extrusion having the width and the height of the nozzle will result in a string with nozzle diameter. A test print has verified this behaviour (here: nozzle width 0.35): the outer extrusion path are piles with rare contact points.
[image]
So for small extrusion width too few material is provided, and since it is proclaimed everywhere to print with nozzle width as default, I believe, that several of the issues in this issuetracker can be improved by an improved extrusion model.
The improved model claims: An extrusion will fill the whole area below the nozzle width. Larger width will be formed by a bulb on the side. 
As a first approach this bulb is assumed to be a circle segment. A graphical expression is like that:
[image]
Driving this approach into an extrema for extrusion of height and width values equal to the nozzle size, the result is like this:
[image]
the extruded piles are touching on large parts and are well arranged.
With this model I could achieve an improved scalability of the extrusion parameters, using different width and height values for a part doesn't affect the dimension or quality (at least the impact is remarkably reduced).
The modified source code can be downloaded here:
https://www.mydrive.ch/browse#/  (guest@hoh61_ext/slic3r)
It replaces the file "flow.cpp" in the directory Slic3r/xs/src/libslic3r.
I would encourage everybody to test this modification, maybe it is worth to be implemented in the next revision of slic3r.

alexrj, 2015-11-06:
Slic3r used that model until mid-2014. Then, after a [discussion](http://forums.reprap.org/read.php?263,273929,page=5) with @nophead, we agreed on the fact that after the extruder moves away, surface tension turns the sides into half circles. The effects on the results were very little.

hoh61, 2015-11-07:
Dear alexjr,

I'm very suprised about the discussion and the resolution I found in slic3r. When you say, the effects were very litte, which effects were reviewed (i didn't found any reference)

First of all the model actually implemented is not bad but limited to the extend I described.
Without knowing this discussion I cumulated all aspects of this discussion into my code.

1) if W - T > nozzle the implemented model is fine, I actually wouldn't know anything better.

2) The actually propagated settings continuously are violating the relation above. Therefore an error is implementing in the material flow, like depicted in my graph and already discussed in the mentioned discussion. I got the impression, that for this condition a special solution was included in slic3r, which is no longer implemented. My proposal of implementation is that the provided material shall be able to fill the layer for 100% in any condition, which is not achieved by the implemented model. This aspect was reviewed in the reference discussion too.

3) In case the  extrusion width is smaller than the nozzle width, no real prediction of profile is possible. In this case the rectangular approximation for the material flow shall be applied to ensure at least the required material flow to fill the gap. In this case the error to the implemented model becomes very large. So I "re-implemented" this condition.

From optical point of view the difference in the model is quite minimal, the impact on the cross sections can be seen above.  The mechanical strength of the part will improve with my implementation providing larger contact areas between the deposited piles.

Finally, from the cross sections I made up to now I never saw the semicircular outer wall, I observed always a circle segment with a bow heigth of 0.3 to 0.4 of layer thickness. That means, that none of the model is perfectly enough to describe the reality, in my implementation I take care that at least the amount of extruded material will fill the foreseen volume.

P.S.: in the documentation a statement was given to the same defaults. In the actual code implementation (in module flow) it is not active. If there is no other point for checking sanity this is a discrepancy to the manual and a nice feature to play around. Default extrusion width smaller than nozzle width will be allowed then!

mrvn, 2015-11-17:
Maybe you are trying to solve the wrong problem. Or looking at it the wrong way. What are you trying to solve? Where is each case in the flow model used?

As I see it extruding less than nozzle width * layer height results in a round string the size of the nozzle that then gets stretched and becomes thinner. For thin layers it is still squished down. Are the squished down sides half circles or some other shape? Well, anyone got a microscope and could look at some cross sections? Personally I think printing a single strand less than nozzle with is  not a good idea. It's easy for the flow to break leaving gaps and wider lines where it resumes.

So point 1: Single (or first) lines are always >= nozzle width. Get a model for that. Get the shape right too because this decides the dimensions of the printed object although the error goes towards 0 with decreasing layer height. Anyone got a 1 mm nozzle to test large layer heights?

Point 2: Printing a second line will expand according to our model in 1 till it hits the first line. It will then try to fill around the shape of the first line. It might not be able to fill all the space leaving some air, mainly at the bottom. The remaining extrusion will then expand away from the first line. By overlapping lines a thinner second line can be added. I believe the outside part of the second line will still follow the model form point 1. Just get a model for how well it fills the gap with the first line.

Point 3: Printing between two lines is like Point 2, except on both sides. The filament will expand to fill whatever space is there. The nozzle size is basically irrelevant (in one direction) since the lines on both sides add extra walls. The filament can expand in the other direction unless those are walled closed too (which they are by the perimeters). So really we only need to know the volume left to fill and how well it can be filled.

Point 4: The above assumed that any overlapping parts will be totally blocked. From what I see with the default 15% infill overlap it actually squeezes some filament into the overlapping part. I'm not sure if that is just the back pressure lifting the nozzle a bit or the filament being compacted and then expanding when the nozzle has moved on but overlaps gain a little height. So that must be modeled to as a correcting factor in the above.

Personally I have to say that slic3r should use more overlap. I often have top/bottom infill where it leaves large, visible gaps, where it can't put a full line anymore. It should add that extra line and then overlap all lines a bit to make all lines equal.

hoh61, 2015-11-19:
What is my problem: I was printing different objects with different settings of line width and layer height. The fine tuning for one gave messy results for the other part. I'm not only looking at the surface condition, which shows not too much difference, but on the mechanical strength. 
Nearly all your comments are o.k., the correct volume calculation is given only for case 1. But with too few material you will end up in a situation where the contact points will be too less, mechanical stability is degraded. You won't see it in devil like marketing printouts but in mechanically stressed parts. That was the reason for my review.
The actually implemented model is not scaling the material properly. I claim to provide an improvement, a better scalability for variable settings, still knowing that it is still not describing the reality. It "only" takes care, that the extruded material fills the available volume completely under any condition.

(As I mentioned, the outer form of the extruded shape is regularly not "!" a semi-circle, but more or less a circle segment with a height of 0.3 to 0.4 times of the layer thickness. In the meantime I found a publication, where the outer shapes of extruded profile were measured. They found no semi-circles but shapes, which could be approximated by a circle segment with a typical height of 0.35 of the layer thickness (surprise). On my way for an understanding of this observation I realized, that the shape of the outer profile is strongly affected by surface tension and the adhesion forces between filament material"!" and nozzle material"!" (minimized filament  surface, see: Young equation). Having this in mind you will have too many variable to model the shape correctly. So I ended up with my personal conclusion that at least the provided material should be calculated correctly.)

Just a word on the infill overlap. In the beginning I was facing an overextrusion at the edge of the infill, leaving the standard value if 15% unchanged. With different settings of this value I got strange effects, which I haven't understood so far. I had no time to analyse this situation in depth, nevertheless, this parameter seems to solve Beelzebub/devil situation at a first glance. My first approach was to reduce the extrusion at the return points of the infill, but now I think this is a dead end. As I said, I have no clear picture yet.

A last remark to your final statement: please review your width setting for the infill. If you're using different width for infill and perimeters, you might be a candidate who will see an improvement with the proposed model. It will scale better.

lordofhyphens, 2016-07-12:
@hoh61 You'd probably be better off implementing this as a pull request and demonstrating its utility as I don't think you've convinced @alexrj at present.

bubnikv, 2016-07-12:
I believe the current slic3r behaves differently from what was described
here originally. The current slic3r places the extrudates next to each
other to fill the space 100% by relying on the effect of the hot nozzle tip
to iron and squish the extrudates to fill the space completely. This could
be confirmed by the extrusion simulator in the plauground branch of the
prusa3d fork. The so called extrusion geometry influences the tolerances of
the outer walls though.

<https://github.com/alexrj/Slic3r/issues/3118#issuecomment-231922329>, or mute
the thread