slic3r/Slic3r #3119: Exporting big files to a SD card is very slow on windows.

DiThi, 2015-11-07:
Try any model that would take many hours to print. It's much faster to export to hard disk and then copy to SD than exporting to SD directly. It seems to happen in both sync and async modes (I'm not sure if I configured async correctly, sync is default for removable devices). Unlike Linux, Windows doesn't seem to have a "flush" mode.

The solution is to buffer e.g. 64K and write when it's full or when the export finished.

alexrj, 2015-11-10:
What operating system are you using?

DiThi, 2015-11-10:
Windows 8.1

ghuser2, 2015-11-17:
I've heard that you should always export to disk first anyways, to prevent certain problems.

DiThi, 2015-11-17:
I realized the problem may be more than what I said. Even if one is writing one whole flash block at a time, the OS will still be updating one FAT index at a time.

The simplest solution is for slic3r to write the whole file in RAM or a temporary folder, then copy to the target device after it has finished.