slic3r/Slic3r #3121: Rotating about an axis is unclear. Which color is which axis?

TTN-, 2015-11-08:
Title is pretty self descriptive. When rotating an item around an axis, on the build platter the axis are indicated by colored, while in the menu they are named x y z. I don't know which one is which.

alexrj, 2015-11-08:
Isn't it pretty obvious actually? Blue is vertical, and the others are XY.

alexrj, 2015-11-08:
Even on your printer I doubt axes are labelled with their letters, but somehow you recognize them :-)

TTN-, 2015-11-08:
"and the others are xy" yep. Which color is which? At the moment I rotate about an axis by trail and error, until I get it right. Which color is x, which one is y? I don't know this off the top of my head.

alexrj, 2015-11-08:
Same orientation as your printer...

TTN-, 2015-11-08:
[image]
Sorry to be persistent. I'd like to illustrate what I'm trying to point out: which axis should I rotate about to get the large hole in the center of the object upward?

a4jp-com, 2015-11-08:
Around X axis... (Color here)
Around Y axis... (Color here)
Around Z axis... (Blue)

Something like this would be nice. I was asking for the same thing in a different thread.

TTN-, 2015-11-08:
Yea, its just something I find myself fumbling with repeatedly...

munzp, 2015-11-08:
The colours of the x, y and z axes are red, green and blue respectively; the x, y and z colours are in the same order as the RGB colour system (this may be coincidental). The axes are also aligned according to the "right hand rule" for coordinate systems where the index finger is +x, the middle finger is +y and the thumb is +z. It is worth noting that there is only one orientation for the three positive axes when defining a coordinate system, if the positions of the x and y axes were swapped then one of the three axes shown would have to become negative to maintain a correct coordinate system.

Incidentally, I found when I was checking the orientation of the coordinate system that the direction of rotation for a positive angle was in the negative direction according to the "right hand fist rule" that is normally used to define rotation direction.

alexrj, 2015-11-08:
@TTN-, it's clearly the Y axis. Think about how the axes are oriented in your printer.

TTN-, 2015-11-08:
Ahhhhh. Of course.

a4jp-com, 2015-11-09:
Even if XYZ = RGB it would still be nice to have the colors listed.

Around X axis... (Red)
Around Y axis... (Green)
Around Z axis... (Blue)

I've been using 3D programs for years and never realized this before.

a4jp-com, 2015-11-09:
I made a pull request here: https://github.com/a4jp-com/Slic3r/compare/master...a4jp-com-patch-1-XYX-RGB-lables?quick_pull=1

I hope the brackets can be used inside the quotation marks.

TTN-, 2015-11-09:
I don't think you've completed the pull request. When you have created it, you will have to push (is that the right word?) it upstream to this repository, as atm it looks like its only going to apply to your fork of the repository.

a4jp-com, 2015-11-09:
Whoops. I'll check again.

a4jp-com, 2015-11-09:
https://github.com/alexrj/Slic3r/pull/3125

koghe, 2015-11-10:
will be great to see axis letter references like blender and others 3d software have, i'm not  colorblind but having a only text reference is not useful for them

[image]

a4jp-com, 2015-11-11:
That would be nice but I guess having the letters always facing the right way around all the time would make coding this a bit of a pain. The I menu update I sent in would only take about 60 seconds to implement at most, unless my coding is wrong.

lordfwahfnah, 2015-11-14:
another tip to memorize http://www.cnczone.com/forums/attachment.php?attachmentid=25504&stc=1&thumb=1&d=1389507392

alexrj, 2015-12-05:
I added nice colored icons:
[image]

closed 2015-12-05 by alexrj

a4jp-com, 2015-12-06:
Thank you very much.

I like this even better than the idea I had ^^

TTN-, 2015-12-06:
Awesome! Thank you.

koghe, 2015-12-06:
Cool, thanks!
i cannot see it on linux+default theme, but i think is a theme problem.

TTN-, 2015-12-07:
fresh download and compile, no icons atm. linux user here too: ubuntu 14.04