slic3r/Slic3r #3122: Frequent (but unpredictable) crashes on OSX

hhhudson, 2015-11-09:
Hello! I'm seeing **very** frequent crashes during slicing on OSX. This probably occurs 80+% of the time for me. One weird part is that they aren't predictable - an object that crashes over and over again will randomly work out of the blue. No idea what the cause is. I have background slicing turned off, because I thought it might help (is didn't).

This might be a duplicate of https://github.com/alexrj/Slic3r/issues/3098, but I don't know enough to actually say that.

Slic3r:
1.2.9 stable

OS:
OSX 10.11.1 (15B42)

Hardware:
MacBook Pro (Retina, 15-inch, Mid 2014)
2.5 GHz Intel Core i7
16 GB 1600 MHz DDR3

Config:
https://gist.github.com/hhhudson/6725129ce4ff4021ff10

Crashlog:
https://gist.github.com/hhhudson/0f65b1b2994373d0a221

File:
http://www.thingiverse.com/thing:53451

hhhudson, 2015-11-09:
Hunh - no crashes if I use the quick slice options at least!

alexrj, 2017-03-17:
@nathanielhudson, you might want to try our new automated builds: http://dl.slic3r.org/dev/
The packaging toolchain is brand new and the executables are very stable now. Feel free to reopen this if you find more crashes!

closed 2017-03-17 by alexrj