slic3r/Slic3r #3123: GCODE generation: voids in some models

vladtcvs, 2015-11-09:
For some models, big voids appears, some parts of a model have only perimeter.

 https://vltc.net.eu.org/owncloud/public.php?service=files&t=fedcdae7216f7d281fdf840194df0110

version 1.2.9, debian.
config.ini:
```
# generated by Slic3r 1.2.9 on Mon Nov  9 03:06:08 2015
avoid_crossing_perimeters = 0
bed_shape = 0x0,170x0,170x170,0x170
bed_temperature = 60
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 70%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_diameter = 3
fill_angle = 45
fill_density = 70%
fill_pattern = line
first_layer_acceleration = 0
first_layer_bed_temperature = 60
first_layer_extrusion_width = 200%
first_layer_height = 0.35
first_layer_speed = 30%
first_layer_temperature = 220
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 1
infill_overlap = 15%
infill_speed = 70
interface_shells = 0
layer_gcode = 
layer_height = 0.4
max_fan_speed = 100
max_print_speed = 80
max_volumetric_speed = 0
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.5
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 40
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 1
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 1
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = aligned
skirt_distance = 6
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 30
small_perimeter_speed = 30
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 60
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 1
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 70
temperature = 220
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0
```

[image]

hoh61, 2015-11-11:
I just played a little bit with your part and found the following effects:
1) I copied the main slicing relevant parameters into my settings (nozzle width, extrusion factors, selected options...) added the STL file and found my computer busy until slic3r crashes.
2) with my own settings I got very soon a complete layer composition.
3) just copied the basic geometry: nozzle width, fill parameter, layer height, advanced: no obvious problems.
4)activated support material: no change.
5) set  support angle 70: generation of large support structures, but no change in layers.
6) switch on thin wall detection option: some silly changes in the infill, but not like yours

I have faced some problems with the thin wall detection option in another structure of mine, so this is generally off in my settings. It was messing up the infill in a similar way like you are reporting it.
Please check if this has an impact on your slicing too.

thirtythreeforty, 2015-11-23:
I have a similar issue with [a very simple model](http://s000.tinyupload.com/download.php?file_id=28453213175237863666&t=2845321317523786366654464) using [this configuration](http://s000.tinyupload.com/download.php?file_id=06689826313245138276&t=0668982631324513827607893).  (This model is originally @PrestonStinson's.)  Only the perimeter wall is generated; no filled base is generated.  First layer:

[image]

The entire sliced model:

[image]

expertmm, 2017-05-16:
This example may be related:
1.3.0-dev 2017-05-09 git
When printing bondi_dwarf.stl from http://www.thingiverse.com/thing:750403
The following settings print hollow (chain mail, which is a separate solid, has paths generated, but actual body, other than solid layers, does not):
* Ultra Quality
* Low Quality
* Medium Quality - Simple Model
The following settings do not print hollow:
* Medium Quality - Complex Model
* Simple Mode
* High Quality

Ultra Quality example:
[image]

lordofhyphens, 2017-06-02:
@expertmm Just to be sure, did Slic3r say it had to repair the model? 

Additionally, we have no idea what your "Quality" settings have in them, please export a separate config.ini (`File->Export Config`) for each one so that we can make sense of your report. 

Thanks.

expertmm, 2017-06-02:
As far as I know I did not change these and they are default presets. Is that not correct?

lordofhyphens, 2017-06-02:
That isn't something we ship. 

You can export each config and zip them up and upload here.

expertmm, 2017-06-06:
[slic3r-expertmm-voids.tar.gz](https://github.com/alexrj/Slic3r/files/1055776/slic3r-expertmm-voids.tar.gz)

expertmm, 2017-06-07:
Just so you  know the gcodecook.py post-processing script's main purpose is to send the gcode to gpx (to convert to x3d), but doesn't modify the gcode.

expertmm, 2017-06-07:
The presets must have come with the prepackaged slic3r that was on my sd card that came with my FlashForge Creator Pro. Since then I had updated slic3r several times and never knew that the presets wouldn't have been there had I not used the version from FlashForge.
[expertmm-SlicerConfigExport.tar.gz](https://github.com/alexrj/Slic3r/files/1059555/expertmm-SlicerConfigExport.tar.gz)
Here are the separate ini files you asked for instead of the whole config folder.