slic3r/Slic3r #3124: Slic3R 1.29, sliced box has no bottom in output

justmyopinion, 2015-11-09:
[image]

[image]

When I slice this (simple box) bottom disappears and I find no settings to get bottom slice output.
Slic3r 1.29/ W7/64

https://www.dropbox.com/s/s9bhrn0mz3x3583/Box_config.ini?dl=0
https://www.dropbox.com/s/fnw7fv715jo8thy/Box_.stl?dl=0

a4jp-com, 2015-11-09:
Printer settings > Horizontal shells > Solid layers... may help

justmyopinion, 2015-11-09:
Ok, found it, there was indeed no dimension created on the bottonplate, but no errors was issued so it was not recognized. Sorry

closed 2015-11-09 by justmyopinion

a4jp-com, 2015-11-09:
^^