slic3r/Slic3r #3126: Holes and gliches on vase mode

thiagopeixoto16, 2015-11-09:
Hello!

I'm here to report two issues with vase mode. 
**1. "Dots"** 
When I turn vase on for this model, it generates some **"dots"** over the inner part of the model, like that:
[image]
It happens only when I raise up the extrusion width (like the example 1mm), lowering it to the nozzle size it disappears.

**2. Holes**
Another issue about the same model, there is some gaps that I called **"holes"**. I've verified and there is a real travel command that skip around half of the loop.
[image]

INI files / STL / Images / gcodes here:
https://www.dropbox.com/s/fjjjvc994qy8xxf/BadVase.zip?dl=0

thiagopeixoto16, 2016-06-29:
I got this error today again on another vase. :cry:

lordofhyphens, 2016-06-29:
:( I don't much care for spiral vase mode as a printing option myself. I've tagged it so it's less likely to slip through the cracks, but time is at a premium and there's, oh, about 500 or so uncategorized old reports still to sort through. 

For someone looking to debug this, probably start seeding print statements around
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/GCode.pm#L353 

Something else that might be handy is to see if it still generates "holes" in the gcode if you turn spiral vase off but set perimeters = 1, top layers = 0 and bottom layers = N, 0% infill (all manual) If it doesn't break there, we can be certain of the STL input.

lordofhyphens, 2016-06-30:
Well, on my build of 1.3.0-dev I didn't see any gaps.
[image]
gcode - [noholes.zip](https://github.com/alexrj/Slic3r/files/340482/noholes.zip) if you want to try printing it.

The blobs look like infill or gapfill according to how Slic3r's 3d preview. Not entirely sure how that's happening. I set gap fill = 0 to see if that was getting used, no dice.

I'm going to add a separate issue for the infill blobs, makes tracking stuff easier.

lordofhyphens, 2016-06-30:
Found the culprit: "Only infill where needed" was checked in your profile and that was causing the blobs.

closed 2016-06-30 by lordofhyphens

lordofhyphens, 2016-06-30:
If you run into problems with new -dev w/r/t holes please open a new issue. AFAIK it's fixed in current -dev now.

PR #3386 (2016-06-30): infill_only_where_needed is incompatible with spiral_vase: enforced.
Also ensured that support is disabled if spiral_vase is on.
Added a secondary check to make sure that clip_fill_surfaces does not run if we're in spiral_vase.

Fixes #3126