slic3r/Slic3r #3127: MakerFarm Printer Z Axis sync

datcomputerguy, 2015-11-12:
There seems to be a problem where when I use the newer Slic3r my Z axis stepper motors get out of sync with each other.

The Manufacturer says that I have to use Slic3r 0.99 because the problem occurs in the newer versions.  I didn't believe them, but it is completely true.  Is there a fix for this already in progress?

I measure a slight problem on my Prusa i3v 12" with a RUMBA board, but I measure a large problem on my Pegasus 8" with a RAMPS board.

markwal, 2015-11-12:
I'm confused. Slic3r just generates gcode that says go to this X Y Z. Your firmware translates that into steps for each stepper motor.  If a new version causes a problem where an old one doesn't, it must be that the firmware can't handle correctly some sequence of moves only produced in the newer version of Slic3r?

First, if there is something to be fixed, it seems mostly likely to be in the firmware.  Second, if something is to be fixed I think more precise information will be needed.  You have more than one Z axis stepper motor?  And some sequence of gcode (could you upload the gcode somewhere and provide a link) and that results in some measurable difference in movement in the two Z axis motors?  How much difference and at one point is that measurable in the linked gcode?

datcomputerguy, 2015-11-12:
I know this sounds weird, seriously it doesn't make sense.  I am using the
newest release of the Marlin firmware for my printers.  My Z axis motors
are connected to the same motor driver, so in theory that shouldn't ever
get out of sync with each other.

The guy who makes the printers at MakerFarm says that it is a Slic3r issues
and that I have to use version .99 of Slic3r.  He said that the new gcode
that the Slic3r files generates makes the Z axis motors move too fast and
apparently causes them to get out of sync with each other, one does more or
less steps than the other over time.

I love the new version of Slic3r and would much rather use it over the .99

When my printer is done printing this long print I will export the 2
different sets of gcode and then print them both on the same printer and
record the results.

jluis, 2015-11-12:
El Dimecres 11 Novembre 2015, a les 17:02:46, datcomputerguy va escriure:

> I know this sounds weird, seriously it doesn't make sense.  I am using the

> newest release of the Marlin firmware for my printers.  My Z axis motors

> are connected to the same motor driver, so in theory that shouldn't ever

> get out of sync with each other.

>

Then is the driver that isn't providing enough juice to move the two motors, If you don't wish to assign to each Z motor its drive, it has advantages as automatic leveling in firmware, you could limit the Z speed and acceleration on the firmware.

Mikk36, 2015-11-12:
Definitely not a bug with Slic3r, it only provides a single Z coordinate, the rest is up to the firmware and/or mechanics (in your case mechanics, because you're sharing the driver for both Z motors).

justmyopinion, 2015-11-12:
SLIC3R is not to blame here.
Z is moved at "travel speed" which can set be very high but this is normally limited by limits set in your FW or EEPROM settings because Z drive  is not able to honour these high settings. 
While X,Y limits could be high, it is quite normal to have a max Z speed of 3-5mm/s for this type of printer.
You may check your max settings for Z in EEPROM or program the limits in FW (overrided by EEPROM!).
If this does not work you may have to increase driver current settings for two Z motors in parrallel.

alexrj, 2015-11-12:
If your printer uses Marlin you can set the max Z speed using this command:

```
M203 Z300
```

Put it in your start G-code.  And please don't use old Slic3r versions, it's totally pointless...

closed 2015-11-12 by alexrj

mrvn, 2015-11-17:
This raises an interesting point though. Speeds are different along different axis. Most notably Z is usually way slower. But movement in X is usually simpler than Y too. Wouldn't it make sense to give  speeds for each axis?

Mikk36, 2015-11-17:
More likely you'd want seperate acceleration values for different axis'. This however needs to be supported by the firmware and if you have issues with that, you're likely to get an answer saying to fix your hardware or use a smaller acceleration value that suits for both axis'.