slic3r/Slic3r #3128: Request: Enabling support for x number of layers only.

Capital-M, 2015-11-12:
Can an option be added to turn off support beyond a specified height or number of layers please?

I have a concave-bottomed holder for a captive nut. This needs to be supported due to the text. 
[image]

But the area where the nut is inserted needs to be left clear.
[image]

I understand that I may be the only person who wishes for this feature, and I could do away with the text, and therfore support would not be necessary.

markwal, 2015-11-12:
You should be able to print that pretty well without support at all. The top of those letters are pretty short bridges.

The request might be interesting for other reasons though.

Capital-M, 2015-11-13:
Without support the letters print in the air.
[image]
[image]

markwal, 2015-11-13:
Ah, that is more clear.

pyrophreek, 2015-11-16:
At the moment, you can solve this by using a modifier mesh. Create a cube that is larger than the object in x and y, but just as tall as the letters, and make sure it is positioned so that when both are imported using the same coordinate references, they overlap. Then go into slicer and load your object, click settings, and load the cube as a modifier mesh. Enable support for the modifier mesh, but disable it for the rest of the print.

pyrophreek, 2015-11-16:
See this post for a slightly more detailed explanation: https://groups.google.com/forum/m/#!topic/3d604/WaY_HFeduPs

lordofhyphens, 2016-11-27:
Actually, you can't disable supports over a modifer mesh currently. I think that this is something that can be done, just that it hasn't been implemented.

@alexrj @bubnikv  any thoughts?

bubnikv, 2017-01-27:
Yes, modifier volumes over supports are on my plate and it has a high priority, I did not start yet though, pulling straw after straw.