slic3r/Slic3r #3129: Estrange resolution slope in Model

Caracol-Studio, 2015-11-15:
Hello I am using Slicer 1.29, I would like to know why the result of  slicing a  stl file composed for  3 pieces,  in the biggest one  the slope  of the pieces is generate  as a scattering surface instead a continuous and smooth as in  the STL file. I  checked in the Netfabb that the pieces  are corrected without corruption in the mesh. as well I have tried to change many parameter getting always the same result. 

I attach link to zip with  the files ( Printing Profile, Stl File, Gcode generate it ). Thanks in advance

https://www.dropbox.com/s/teqvzqaezveorpc/slic3r_error.zip?dl=0

[image]
[image]
[image]

ghuser2, 2015-11-17:
Probably because a slicer needs to have separate layers on the slope. The stl does not.

hoh61, 2015-11-19:
or in other words: any slope must be approximated by steps. The smaller the layer height, the smaller the step  becomes (by the cost of printing time), the better is the approximation to the .stl file.

p.s.: I did a short review of the GCODE file:why did you chose 0.4mm for the first layer (= nozzle width). Furthermore several extrusion width were selected smaller than the nozzle width, which will generate significant errors in material extrusion in slicer. My personal suggestion for the settings would be: first layer = extruded layer = 0.2mm, extrusion width default: 0.45mm, first layer width 0.6mm. Maybe you have tried this already, A further improvement would be to reduce the layer height to 0.1mm (of possible with your hardware).