slic3r/Slic3r #3131: Feature Request: Multiple configs for a single slice, with select layer ranges using Config INI Section Names

h3rb, 2015-11-16:
It would be extremely useful for me to be able to say "use this config for these layers", where I could specify infill (and other options) for specific layers  .. this feature is probably an enhancement of the "every nth layer" feature.

ie:

; For setting special parameters on the first layer that override the 'master settings'

[special_settings_for_layer_1]  ; this is a "config section" name
start_layer=1
end_layer=1
... settings here

; For setting special params on the 2nd-10th layer:
[special_settings_for_layers_1_10]
start_range=2
end_range=10
... settings here

etc

ghuser2, 2015-11-17:
Isn't this how modifiers work?

h3rb, 2015-11-17:
Can someone show me the documentation for modifiers?  I'm not yet convinced I have the full control I'm hoping for.  What are the limits of mesh modifiers?  CAn you have multiple mesh modifiers interacting with the same layer?  Can you have any mesh shape or only cubes?  Do they have to be 2d extruded along the Z or can they be a sphere for example?

lordofhyphens, 2016-11-27:
See #3590 and http://slic3r.org/blog/modifier-meshes

closed 2016-11-27 by lordofhyphens