slic3r/Slic3r #3132: Feature Request: Allow the gcode to be commented to mark layer end / layer transition / layer start

h3rb, 2015-11-16:
In the .gcode, have slic3r automatically add comments that mark these areas, so if you wanted to slice up a file and reengineer it after the slice occurs, you would know where the hell one layer starts and where one layer ends

markwal, 2015-11-16:
I can already make slic3r do all this (unless I'm misunderstanding the request).

In "Print Settings"."Output Options" there is a checkbox "Verbose G-code".  If you turn that on it adds comments to every line that says what it is for (infill, for example).

Also, since you can put whatever custom gcode you want (including comments and there are variable expansions even) you can have the layer transition marked however you want: "; new layer"

pm3d, 2015-11-17:
That may be too verbose, perhaps there could be a less intensive setting that marks the 'major moments' only?

markwal, 2015-11-17:
Shouldn't matter much if you're post processing anyway. You could strip them all out during the post process so you don't have to send them to the printer.

markwal, 2015-11-17:
And like I say, you can do layer change by itself if you like.

pyrophreek, 2015-11-17:
+1
this would be useful for a number of things, including post-processing scripts.  Also, having a comment at the start of each perimeter could be helpful to distinguish inner, outer, middle etc.

pyrophreek, 2015-11-17:
@h3rb 
as far as layer range based settings goes, modifier meshes actually serve this purpose quite well.  One way you could make it easier though, would be to introduce a "create modifier cube" button, which you could use to generate a simple modifier mesh cube with given size and position so that you wouldn't have to create it in a separate program.

h3rb, 2015-11-17:
"modifier meshes actually serve this purpose quite well."  .. so how would you say "do 100% infill no perimeters" on an area using this method?  Also does this work for more complex modifier meshes?  It's just not clear to me how to associate "this region enclosed by this mesh" with "this set of config parameters"?

pyrophreek, 2015-11-17:
Here is a tutorial I wrote for a friend:

"You can make modifier meshes using Slic3r to set specific print parameters for different regions in a single part.  For your example, you might want a solid layer (or more) at 5mm from the bottom, then another at 12 etc.  Import your design into a modelling or cleaning program (netfabb free is great for this, you can import part, move it to origin, then make a cube primitive and position it), then add a cube that intersects your part at the desired height, with the desired thickness.  
[image]
Export the model and the cube as separate stl files (this is to ensure they share the same coordinate reference system), then load the part into slicer, and click the settings button (or right click part and select settings). 
[image]
Now load the cube(s) as modifier meshes, and change the infill on it(them) to 100%.
[image]
"

Hope this helps

h3rb, 2015-11-17:
Can all of this be done by command line?

pyrophreek, 2015-11-17:
not as far as I know, but I have never tried.  Normally I want to visually confirm that things are properly aligned anyways.

h3rb, 2015-11-17:
yes but i would be generating the modifiers mathematically

pyrophreek, 2015-11-17:
That sounds interesting, but honestly I have no idea if the command line interface supports modifiers.  Would be a great way to do it though, as you might be able to just use a simple 1cm*1cm cube as a standard modifier, then import it for each height range you want, just scale and translate to apply it to the desired area.

h3rb, 2015-11-17:
right, or through configs, how to say in a config "use this modifier, with
these settings" ...

lordofhyphens, 2016-07-27:
We have before/after layer change gcode, so you could add your markers there in the interim.

bubnikv, 2016-10-26:
> That sounds interesting, but honestly I have no idea if the command line interface supports modifiers. 

It does, through the AMF files.