slic3r/Slic3r #3133: Feature Request: "Create Modifier Cube" button in part Settings

pyrophreek, 2015-11-17:
Often it is desirable to change settings at various heights, or on different sides of a part.  While it is relatively easy to create a modifier mesh to allow the application of different settings to different regions of a part, it can become tedious if a lot of different meshes need to be made.  In addition, this is only easy if the user is familiar with 3D modelling software, and even then it can be somewhat confusing if the part is not imported and aligned with the created modifier mesh(es).  

I propose that for these simple modifier meshes, it would be much easier to allow slic3r to create them by scaling and translating a simple unit cube.  

Current state:
[image]

Proposed implementation:
Either add a button, or make the "Load Modifier" button a dual use button (drop down menu on the right side to select which) named "Create Modifier Cube" with the tooltip: "Creates a cube with given dimensions and position to be used as a modifier mesh for the part."  which will add a modifier cube which can then be positioned and scaled in 3D.  

Corollary: 
Allow the scaling and re-positioning of loaded modifier meshes to allow for correcting mis-aligned meshes, and for reusing modifier meshes for multiple areas of a part.

lordofhyphens, 2016-11-27:
Done. See #3590 I instead let the user slide the mesh around as they want.

alexrj, 2016-12-18:
@lordofhyphens's feature was merged. Yeah!

closed 2016-12-18 by alexrj