slic3r/Slic3r #3135: allow changing filament settings at n layer height (M600)

robseee, 2015-11-23:
Hi,

since Marlin supports M600 filament-change command, it would be handy to be able to change the filaments settings (temperature / diameter / multiplier) at a specific layer height.

M600 gcode support from the slic3r GUI would be nice too

Thank you

lordofhyphens, 2015-11-23:
This can be handled with a post-processing script, or just editing the
gcode for a single object. For multiple objects or vase mode things get a
little harder.

robseee, 2015-11-23:
how can you change the multiplier with a post-processing script ?

lordofhyphens, 2015-11-23:
Make use of m221, which changes the flow extrusion rate.

lordofhyphens, 2015-11-23:
There might also be something you could do with the modifier meshes.

lordofhyphens, 2016-06-28:
Actually, the other semitrivial way you can make this work is to tell slic3r you have multiple extruders and change the toolchange gcode to insert M600. Do a find/replace in the gcode to remove the toolchange command itself (basically everything but T0).

lordofhyphens, 2016-11-27:
Okay, with #3590 and toolchange gcode, tell Slic3r you have 2 extruders, define a modifier mesh to cover the area you want to change to, and put M600 in the toolchange gcode. 

Should work.

closed 2023-09-23 by robse-git