slic3r/Slic3r #3136: Can't compile new code for the developer version.

a4jp-com, 2015-11-23:
MANIFEST appears to be out of sync with the distribution
cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\design/.cpanm/work/1448302360.20728
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.13
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/slic3r-4/xs
Checking if you have Module::Build 0.38 ... Yes (0.4212)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Running Build.PL
Configuring C:/slic3r-4/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_DIR environment variable:

    BOOST_DIR=/path/to/boost perl Build.PL

N/A
! Configure failed for ./xs. See C:\Users\design\.cpanm\work\1448302360.20728\build.log for details.
The XS/C++ code failed to compile, aborting

markwal, 2015-11-23:
See https://github.com/alexrj/Slic3r/issues/3117

a4jp-com, 2015-11-23:
Can you add these two links to the top page (https://github.com/alexrj/Slic3r)?

https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows
http://www.boost.org/

-------------------------------
Also instructions.

Copy the Boost folder to the C drive of your PC then open command prompt and type in:

cd..
cd..
cd boost_1_59_0

When you get to the Boost directory type in:

bootstrap.bat mingw

then

b2 toolset=gcc

You should see this:

The Boost C++ Libraries were successfully built!

The following directory should be added to compiler include paths:

    C:\boost_1_59_0

The following directory should be added to linker library paths:

    C:\boost_1_59_0\stage\lib

----------------------------

What do I type in now?

alexrj, 2015-11-23:
`perl Build.PL`

alexrj, 2015-11-23:
Actually, the linked instructions recommend to put boost in C:\dev, but that shouldn't create problems.

a4jp-com, 2015-11-24:
I did and it compiles but now it will crash when trying to load a model.

[image]

a4jp-com, 2015-11-24:
I can reload the boost files into C:\dev if it is better as a test.

a4jp-com, 2015-11-24:
Oh... a different file opened but this file worked before.

alexrj, 2015-12-19:
Can this be closed?

a4jp-com, 2015-12-19:
Okay ^^

closed 2015-12-19 by a4jp-com