slic3r/Slic3r #3137: [request] Show approximate print time in info

Lord2Vader, 2015-11-24:
Would be nice to get a print time shown in the info.

With repetierhost i get that, but not in slic3r itself.
I upload a gcode to octoprint and then it shows like 24h print. Would be nice to know before uploading ;)

mrvn, 2015-11-24:
Include the filament length / volume and estimated weight too. When I print at the local fablab (makerspace) I have to pay by weight so it would be nice to know beforehand.

pyrophreek, 2015-11-24:
I would suggest integrating this into #2979.  This would be best if you could slice the file in Slic3r without exporting it to view filament usage, time, etc.  Then if it is too long, or too much filament, you could adjust settings prior to uploading/saving the gcode.

claell, 2016-08-19:
Any update on this?

lordofhyphens, 2016-08-19:
The filament length is already in the gcode; grep for "; filament used = " in the output gcode.

@kliment put an estimator into #1334. If someone wants to cook up a clean planner/estimator and submit it as a PR, I'll at least look at it and review it for a PR. I know I have no interest in working on such a thing myself.

closed 2016-08-19 by lordofhyphens