slic3r/Slic3r #3138: Skirt do not attaching to the object (light brim tentative)

thiagopeixoto16, 2015-11-24:
Hello!

I'd like to do something the people do at Cura, that is set skirt to 0 and have a brim for the object border only, however when I set that to 0 I get something like the images below.
I have printed that to make sure this is not just a visualization mistake.
Is it possible to 0 really means attached or allow negatives values for that parameter?

Thanks!

[image]
[image]

a4jp-com, 2015-11-24:
I see what you mean. 0mm is not 0.

pyrophreek, 2015-11-24:
have you tried setting it to 0.01? I think setting it to 0 may  be interpreted as just using the default, rather than actually 0.

Sebastianv650, 2015-11-24:
I'm maybe missing something, but why you don't use the brim feature for a brim?

pyrophreek, 2015-11-24:
if you are printing something with very convoluted outer perimeter (such as a gear), it can be difficult to remove the brim from inside the concave areas (ie. between teeth).  In this case, it would be almost as effective to prevent warping to attach only the tips of the teeth to the brim, while making it much easier to remove.  Or at least this is my experience.

a4jp-com, 2015-11-25:
Even if the distance is set to 0.01mm there is still a gap that is much larger.

[image]
[image]

I agree with @pyrophreek. If this setting of 0mm or 0.01mm etc can connect at 0mm, it would be wonderful. I would be able to make bases for people without having to model anything very quickly which is very useful. A brim is only 1 layer thick so this skirt fix would allow so many new things to be created quickly. If it's okay could you please adjust the code here @alexrj .

thiagopeixoto16, 2017-05-27:
I think we can close this due the new Brim distance on this [commit](https://github.com/alexrj/Slic3r/commit/af46db320f820cc68e692cccf2f83890f4ee1596)

closed 2017-05-27 by thiagopeixoto16