slic3r/Slic3r #3139: Another bridging issue

BurntNoodles, 2015-11-25:
Hello,

(using Slicer 1.2.9, windows 7 x64)
I have what should be a pretty straightforward part to bridge, but it looks like the bridge detection is having some issues. Here is the layer before the bridge:

[image]

Here is the first bridge layer:

[image]

The specific issues are that on one bridge (the bottom in the image), it chooses the longest direction to bridge; the bridge on the other side (which should be symmetric?) is at some peculiar angle.
From hunting around on here, it looks like this has cropped up before, and sometimes makes a re-appearance in the main slic3r branch- maybe this happened again?

A maybe related issue, the next layer after the bridge doesn't merge the solid infill into one region:

[image]

Thanks for any help!

Here is the .stl file and config:
https://www.dropbox.com/s/2g3vaxrwnons29k/BridgingIssues%2011-25-2015.zip?dl=0

lordofhyphens, 2017-03-27:
Unfortunately Dropbox has killed your test case. Can you please reupload them here?

lordofhyphens, 2017-03-27:
Alternatively, could you test with a recent build from http://dl.slic3r.org/dev?

VanessaE, 2019-01-12:
This may have worked properly at some point since this issue was filed, but I can confirm a substantially similar issue with commit 748dd3c23a31e34a155c325f281d21b7cd516c72 (2019-01-05) also.

Example:

[image]

I would have expected the bridging infill to be drawn parallel to the perimeters, since this is a rectangular gap.

Model:  [y_tensioner_with_internal_nut_retainer_-__M3_bearing_screw.stl.zip](https://github.com/slic3r/Slic3r/files/2752536/y_tensioner_with_internal_nut_retainer_-__M3_bearing_screw.stl.zip)
Config: [config-20190112.ini.zip](https://github.com/slic3r/Slic3r/files/2752540/config-20190112.ini.zip)

You can also see something similar happen with a Benchy, when it starts bridging over the embossed lettering on the bottom.