slic3r/Slic3r #3140: proble with support

jfpion, 2015-11-26:
I have uncover a pb in the support detection algorythm (i think !) 
the better to understand it is a picture:
[image]
if i specify 50° the support is ok bu with 45 no support , but the lower end of the part don't touch the bed, so the print start in the air...

not a very blocking problem, but ...

mrvn, 2015-11-30:
Can you post the stl file?

a4jp-com, 2015-11-30:
Have you tried adding support to the model through a CAD program? Where is your config file and model?

jfpion, 2015-11-30:
i took the stl file from my cad with sketchup , i made sure it was correct with netfabb and process it with slic3r
here the stl file (rename it as .stl) and the config.ini (rename it too)
[config.txt](https://github.com/alexrj/Slic3r/files/47888/config.txt)
[support batterie MB.txt](https://github.com/alexrj/Slic3r/files/47887/support.batterie.MB.txt)
in i ask for support from 60° it work fine
thank you