slic3r/Slic3r #3141: Crashes on loading STL - models

Luginets, 2015-11-28:
Often crashes on Windows 10 when loading models to platter (especially often when loading multiple models)
You can reduce chance to crash if switch to 2D preview.
[image]

a4jp-com, 2015-12-15:
3D file and config please.

Luginets, 2015-12-15:
It is not depend from  exact file. Slic3r can crash on  any file, more you load at once - more chance to crash. I just successfully load mentioned 2 files to slic3r, but when i try to load them with next file  it's crashes. then i load all these files successfully. I start to suspect my videocard  memory. but all tests proves - it's ok. Previous stable version of slic3r (with 2D platter) works fine.

I can't upload  the files, but you can download them from github: 
https://github.com/reprappro/Mendel-3/tree/master/STLs
https://reprappro.com/documentation/mendel-3/

[Slic3r_config_bundle.txt](https://github.com/alexrj/Slic3r/files/62886/Slic3r_config_bundle.txt)

a4jp-com, 2015-12-15:
Maybe the STL files are broken. I can open any file I make on my PC. 

Please tell me if this crashes your program:

http://a4jp.com/3d-files/bugs/a4jp.stl

a4jp-com, 2015-12-15:
I tried 1010.STL and it crashed the program.

Luginets, 2015-12-15:
this file open without program.
[image: Встроенное изображение 1]

The main bug is - as i understand crash not depend from file. I can open it
fine once and get  a crash on  next open.