slic3r/Slic3r #3142: Slic3r 1.2.9 for OSX El Capitan crash

RichardNg, 2015-11-28:
Happens when drag and drop a .stl to Slicer, but works if use the Add button

MechaNickW, 2015-12-16:
I have this same issue, except it happens for multiple things in Slic3r on El Capitan. I had lots of regular crashes with Yosemite and Mavericks too, though, with 1.2.9. I refuse to give up using Slic3r, since it's my favorite slicing program, but would love to have a smoother experience within the OS X environment if possible. If anyone wants crash reports, I'm happy to provide them regularly. Big thanks to the creators and those working hard to support this fantastic software!

RichardNg, 2015-12-17:
One more thing. When I load a large model. Slicer displays the entire object. I wait till the object to be fully loaded then export a gCode file. Then I load it to Pronterface to print. only part of the object was printed, seems like the gCode was incomplete.

RichardNg, 2015-12-19:
Just want to update on my new finding. When I reversed to Slicer 1.1.7. It doesn't crash as often. Which leads me to think that it was the 3D vies may be the problem. The 3D view is nice tho 😀

MechaNickW, 2015-12-20:
I don't remember 1.1.7 crashing as much, either, before the 3D view was instituted. There was a marked improvement in slicing speed and finished objects, though, between 1.1.7 and 1.2.9, that I would be loathe to give up. I've had okay luck ensuring that I change all of the other parameters like filament diameter, bed offset, infill, etc. before loading the model. When you change these things, the progress bar for loading the model shows up in the lower right-hand corner, so it seems that it's re-slicing or adjusting in some way that tends to crash the program more. My process now is to modify all parameters, load the model by clicking "Add" and finding the file in Finder, then send to OctoPrint/save as Gcode. This causes crashes the least amount of time, but it still happens.

alexrj, 2017-03-18:
We now have nice stable automated builds, which are much better on OS X. Get them at http://dl.slic3r.org/dev/

closed 2017-03-18 by alexrj