slic3r/Slic3r #3143: Feature Request: configurable skirt layer height

mrvn, 2015-11-30:
The skirt uses the same layer height as the object. When I use a thin layer height this means the skirt gets very thin too. This has problems:

1) It needs many loops of the skirt to extrude the minimum amount (7.5mm in my case).
2) The skirt gets hard to remove. you have to use a knife to peel which easily damages the print plate or tape on it.
3) Because the skirt is so thin is easily breaks while peeling off, making point 2 even worse. A slightly uneven print plate causes the skirt to be unevenly thick as well, making some parts even thinner.

It would be nice if one could set a layer height for the skirt to reduce print time and ease peeling it off.

alexrj, 2015-12-19:
That would be very complicated and would defeat one of the main purposes of skirt, which is purging the nozzle before each layer. Aren't you using a thicker First Layer Height?

bubnikv, 2016-06-24:
The skirt has a dual purpose actually. Not only it is used to clean and fill the nozzle, but also to verify the bed leveling around the object to be printed. Therefore it is good to make a full circle around the object.

I agree with the arguments given by @mrvn. Printing PLA on Ultem film works better with a thin first layer, if you are able to control the distance of a nozzle to the bed surface accurately, either by having an accurate flat print surface, or by a mesh bed leveling. Removing a very thin layer of ABS from Utem foil is quite a challenge.