slic3r/Slic3r #3144: Install slic3r on opensuse leap 42.1

sdloo, 2015-11-30:
Hi alexrj and team

i've got the problem, when i'm trying to install slic3r as described. maybe you can help me. please have look at the log.

First my perl-V output.

Summary of my perl5 (revision 5 version 18 subversion 2) configuration:
   
  Platform:
    osname=linux, osvers=4.1.10-1-default, archname=x86_64-linux-thread-multi
    uname='linux cloud105 4.1.10-1-default #1 smp preempt mon oct 12 11:24:48 utc 2015 (67a24e6) x86_64 x86_64 x86_64 gnulinux '
    config_args='-ds -e -Dprefix=/usr -Dvendorprefix=/usr -Dinstallusrbinperl -Dusethreads -Di_db -Di_dbm -Di_ndbm -Di_gdbm -Dd_dbm_open -Duseshrplib=true -Doptimize=-fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -g -Wall -pipe -Accflags=-DPERL_USE_SAFE_PUTENV -Dotherlibdirs=/usr/lib/perl5/site_perl -Dinc_version_list=5.18.0/x86_64-linux-thread-multi 5.18.0 5.18.1/x86_64-linux-thread-multi 5.18.1'
    hint=recommended, useposix=true, d_sigaction=define
    useithreads=define, usemultiplicity=define
    useperlio=define, d_sfio=undef, uselargefiles=define, usesocks=undef
    use64bitint=define, use64bitall=define, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cc', ccflags ='-D_REENTRANT -D_GNU_SOURCE -DPERL_USE_SAFE_PUTENV -fno-strict-aliasing -pipe -fstack-protector -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64',
    optimize='-fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -g -Wall -pipe',
    cppflags='-D_REENTRANT -D_GNU_SOURCE -DPERL_USE_SAFE_PUTENV -fno-strict-aliasing -pipe -fstack-protector'
    ccversion='', gccversion='4.8.5', gccosandvers=''
    intsize=4, longsize=8, ptrsize=8, doublesize=8, byteorder=12345678
    d_longlong=define, longlongsize=8, d_longdbl=define, longdblsize=16
    ivtype='long', ivsize=8, nvtype='double', nvsize=8, Off_t='off_t', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='cc', ldflags =' -L/usr/local/lib64 -fstack-protector'
    libpth=/lib64 /usr/lib64 /usr/local/lib64
    libs=-lm -ldl -lcrypt -lpthread
    perllibs=-lm -ldl -lcrypt -lpthread
    libc=/lib64/libc-2.19.so, so=so, useshrplib=true, libperl=libperl.so
    gnulibc_version='2.19'
  Dynamic Linking:
    dlsrc=dl_dlopen.xs, dlext=so, d_dlsymun=undef, ccdlflags='-Wl,-E -Wl,-rpath,/usr/lib/perl5/5.18.2/x86_64-linux-thread-multi/CORE'
    cccdlflags='-fPIC', lddlflags='-shared -L/usr/local/lib64 -fstack-protector'

Characteristics of this binary (from libperl): 
  Compile-time options: HAS_TIMES MULTIPLICITY PERLIO_LAYERS
                        PERL_DONT_CREATE_GVSV
                        PERL_HASH_FUNC_ONE_AT_A_TIME_HARD
                        PERL_IMPLICIT_CONTEXT PERL_MALLOC_WRAP
                        PERL_PRESERVE_IVUV PERL_SAWAMPERSAND
                        PERL_USE_SAFE_PUTENV USE_64_BIT_ALL USE_64_BIT_INT
                        USE_ITHREADS USE_LARGE_FILES USE_LOCALE
                        USE_LOCALE_COLLATE USE_LOCALE_CTYPE
                        USE_LOCALE_NUMERIC USE_PERLIO USE_PERL_ATOF
                        USE_REENTRANT_API
  Built under linux
  Compiled at Oct 25 2015 05:49:53
  @INC:
    /usr/lib/perl5/site_perl/5.18.2/x86_64-linux-thread-multi
    /usr/lib/perl5/site_perl/5.18.2
    /usr/lib/perl5/vendor_perl/5.18.2/x86_64-linux-thread-multi
    /usr/lib/perl5/vendor_perl/5.18.2
    /usr/lib/perl5/5.18.2/x86_64-linux-thread-multi
    /usr/lib/perl5/5.18.2
    /usr/lib/perl5/site_perl
    .
and now the output after do perl Build.pl

App::cpanminus is up to date. (1.7039)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.05)
--> Working on Encode
Fetching http://www.cpan.org/authors/id/D/DA/DANKOGAI/Encode-2.78.tar.gz ... OK
Configuring Encode-2.78 ... OK
--> Working on Encode::Locale
Fetching http://www.cpan.org/authors/id/G/GA/GAAS/Encode-Locale-1.05.tar.gz ... OK
Configuring Encode-Locale-1.05 ... OK
[... 122 lines cut ...]
Configuring /home/OWNER/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.18)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have Module::Build 0.3601 ... Yes (0.4003)
Checking if you have Test::More 0 ... Yes (0.98)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.19)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /root/.cpanm/work/1448893996.13133/build.log for details. Retry with --force to force install it.
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~                                                                                                                                                                                                                                                                  
~

kdupke, 2015-12-02:
just run the latest Slic3r on openSUSE 42.1 leap.

I was missing boost, and had to remove /usr/lib/perl/site-perl to make it running.

sdloo, 2015-12-04:
hi kdupke

where did you remove this?

kdupke, 2015-12-04:
/usr/lib/perl5/site_perl/

Attention: If you run other perl programs these might have placed cpan-builds here, too.

alexrj, 2017-03-17:
Looks like something is missing from the compilation output, as I can't see the actual error where it stopped.
@sdloo, I'm closing this one as I guess it's pretty old but feel free to reopen if you need further assistance. We applied a LOT of fixes and improvements so I expect this to work better now.

closed 2017-03-17 by alexrj