slic3r/Slic3r #3145: Slic3r locks up with this model

chakani, 2015-11-30:
Slic3r locks up and freezes with the attached model "Pierre5.stl". 

I can move it around a bit, but Slic3r soon goes catatonic.

Clicking on "Split" makes Slic3r unresponsive, have to kill the task.

chakani, 2015-11-30:
Huh? This is a 3D printing forum and you don't support uploading .STL or .ZIP file types?

Rename the attached file from "Pierre5.jpg" to "Pierre5.STL" to fool GitHub.
[image]

gege2b, 2015-11-30:
Hi
your model is non-manifold (more than 3 thousand errors shown by slic3r)
it cannot make miracles, so it hangs on my PC too ^_^
try to use netfabb or meshlab to cleanup your model

file upload restriction is due to github, witch is not really a forum, and not only for 3d printing

chakani, 2015-11-30:
Hi Alex -
      
      Thank you for your kind reply.
      
      I'm just learning how to generate STL files. I'm creating them
      with a program I wrote myself.
      
      I have installed Netfabb, but I don't understand what exactly is
      wrong with my STL files. Netfabb shows some shapes as being in
      error, but doesn't say exactly what's wrong with them.
      
      Thanks again,
      
      Pierre
    
    On 30 Nov 2015 4:41 PM, gege2b wrote:
    
    
      Hi
        your model is non-manifold (more than 3 thousand errors shown by
        slic3r)
        it cannot make miracles, so it hangs on my PC too ^_^
        try to use netfabb or meshlab to cleanup your model
      file upload restriction is due to github, witch is not really a
        forum, and not only for 3d printing
      —
          it on GitHub.
      
        
          
          
        
        
      
    
    
      Progeny Genealogy Inc. 902–681–3102
      We help you tell the Story of Your Family ™
      progenygenealogy.com

gege2b, 2015-12-01:
Hi (I'm not alex :)

According to meshlab, there are a lot of duplicate facets, thats certainly why slic3r hangs, 
Anyway I still don't know how well to use meshlab, so maybe I'm missing some infos

upload your model to https://netfabb.azurewebsites.net/ (need a windows live ID...) and try with the repaired model

lordofhyphens, 2015-12-01:
It isn't a 3d printing forum platform, github is for software. Just use a
file locker site like 2shared, renaming files can cause issues with github
interpreting things weirdly and changing the file.

chakani, 2015-12-02:
My sincere apologies! My model was pure
      crap  :o)  It's my first effort, and there were thousands of
      duplicated facets.
      
      Thanks to your tip, I was able to see them clearly in Netfabb,
      "Repair" mode. 
      
      I have found many other flaws, and am busy fixing my program.
      
      Thanks again,
      
      Pierre
      
      On 30 Nov 2015 4:41 PM, gege2b wrote:
    
    
      Hi
        your model is non-manifold (more than 3 thousand errors shown by
        slic3r)
        it cannot make miracles, so it hangs on my PC too ^_^
        try to use netfabb or meshlab to cleanup your model
      file upload restriction is due to github, witch is not really a
        forum, and not only for 3d printing
      —
          it on GitHub.
      
        
          
          
        
        
      
    
    
    
Kyber Consulting
Disaster Recovery Planning
902-681-3102

lordofhyphens, 2017-03-28:
Closing. Please retest with current -dev version (http://dl.slic3r.org/dev) if you still think this is a problem.
Github now permits uploading .zip files.

closed 2017-03-28 by lordofhyphens