slic3r/Slic3r #3146: Bridging is not active when doing solid fill over sparse fill

hippich, 2015-12-04:
So I am playing with attached test model to test bridges mostly - https://thingiverse-production-new.s3.amazonaws.com/assets/b0/f3/01/ee/af/ultimate_calibration_test.STL

What I found that Slic3r uses regular solid fill pattern when filling over 5% infill. This results in broken solid infill, since: speed to to fast, often there is nothing to anchor to.

In my opinion, to get best adhesion, it should treat solid infill area as any other bridge. I.e. it first should do perimeter and then run bridge infill with appropriate speed/flow/width/etc.

[image]

[ultimate_calibration_test-2015-12-3-19-48.gcode.txt](https://github.com/alexrj/Slic3r/files/51590/ultimate_calibration_test-2015-12-3-19-48.gcode.txt)

[ultimate_calibration_test.STL.txt](https://github.com/alexrj/Slic3r/files/51591/ultimate_calibration_test.STL.txt)

mrvn, 2015-12-08:
I disagree on the speed. The speed ins't the problem. As I see it there are 2 solutions here:

1) extend the solid fill until the endpoint rests on something. That means either it hits a perimeter and the 15% overlap will hold the infill in place, or it hits the 5% infill and rests on that. For the infill cases it should extend a bit beyond the infill line so it rests solidly on it.

2) Create support lines in the layer(s) before. What I mean there is that it first should approximate the border of the solid fill that hangs in the air with a polygon(s) of few lines (approximate a circle with e.g. a larger octagon). Then it should extend each side of the polygon(s) till they rest on something and print those lines. Note: The lines would overlap at the corners of the polygon(s). The solid fill should then also be extended to rest on the polygon.

In both case the idea is to not have any endpoints hanging in the air and fall down. Option 1 is probably the best with high infill. Option 2 for sparse or concentric infill. Slic3r should probably do a mixture of the two. Try to find a supporting infill line near the solid fill border and create a support line when none is near.

clough42, 2016-01-01:
I don't know if my situation is the same, but I'm seeing similar-looking failures, with unsupported extrusions ending in open space, like this:

[image]
[image]
[image]

Models and config files:

[ShroudBridges2.zip](https://github.com/alexrj/Slic3r/files/75855/ShroudBridges2.zip)

Sorry for the blast.  I'm attaching these examples to two or three open issues that look like they might be related instead of opening a new issue.