slic3r/Slic3r #3147: Multiple extruders  filamentsettings bed first layer

alke, 2015-12-05:
Hello
When I print with extruder 2, bed first layer temp is taken from extruder 1 filament settings. Pulled 2015-12-05 on reprap mendel tricolour installed on ubuntu 14.04

Regards Kent

lordofhyphens, 2017-03-27:
Testcase attached--the provided configuration sets all extruder values to 2 @alke is this what you were looking for?
[3147-testcase.zip](https://github.com/alexrj/Slic3r/files/871299/3147-testcase.zip)

I don't think toolchange is taken into account before start gcode (specifically, temperature) is added if the default expansion is used. 

A workaround is to maintain a separate printer profile that just has the second extruder on it (and issues T1 in its start gcode).

lordofhyphens, 2017-03-27:
A complicating factor here is that the number of extruders used for a print isn't explicitly known ahead of time; Slic3r assumes that extruder 0 is the "main" extruder and that the bed temp chosen for its filament is compatible. I tried to work out that from the choice of the different _extruder config options, but while they exist in $gcodegen->config, the bed temperature isn't broken out.

To do all this just to support having a single input profile (rather than two) is more work than it's worth for me.