slic3r/Slic3r #3148: Can't compile: Error building xs.dll file from 'src\libslic3r\GUI\GUI.cpp

a4jp-com, 2015-12-06:
Newest developer version: 1.2.10-dev

error building xs.dll file from 'src\libslic3r\GUI\GUI.cpp' at C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\Users\design\.cpanm\work\1449393258.16120\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

C:\slic3r-4>

alexrj, 2015-12-06:
Try now. If it doesn't work, post the contents of build.log as the error message says.

a4jp-com, 2015-12-07:
I downloaded the latest version 5 minutes ago and couldn't compile this time either:

error building xs.dll file from 'src\libslic3r\GUI\GUI.cpp' at C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\Users\design\.cpanm\work\1449462128.15088\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

-------------------
Here is a copy of the build.log:

cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\design/.cpanm/work/1449462128.15088
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.13
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/slic3r-4/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4212)
Configuring C:/slic3r-4/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have Module::Build 0.3601 ... Yes (0.4212)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\Users\design\.cpanm\work\1449462128.15088\build.log for details. Retry with --force to force install it.

a4jp-com, 2015-12-07:
I also tried 

perl Build.PL --force 

but that didn't help.

kefir-, 2015-12-07:
Installing ./xs failed. See C:\Users\design.cpanm\work\1449462128.15088\build.log for details.

I think the contents of this file is what alexrj needs.

mihai9, 2015-12-07:
may not be the same thing but after an system update i get lots of errors starting with :

```
In file included from /usr/include/c++/6.0.0/random:49:0,
                 from /usr/include/c++/6.0.0/bits/stl_algo.h:66,
                 from /usr/include/c++/6.0.0/algorithm:62,
                 from /usr/include/boost/polygon/isotropy.hpp:22,
                 from /usr/include/boost/polygon/polygon.hpp:12,
                 from src/libslic3r/Point.hpp:128,
                 from src/libslic3r/Config.hpp:13,
                 from src/libslic3r/Flow.hpp:5,
                 from src/libslic3r/Layer.hpp:5,
                 from src/libslic3r/Layer.cpp:1:
/usr/include/c++/6.0.0/bits/random.h:271:17: error: macro "seed" passed 1 arguments, but takes just 0
       { seed(__s); }
```
bash-4.3$ perl -v

This is perl 5, version 22, subversion 0 (v5.22.0) built for x86_64-linux-thread-multi

a4jp-com, 2015-12-07:
What I posted was the contents of the file @kefir- . 

I can send the actual file or anything else needed if it helps.

mihai9, 2015-12-07:
my build.log :

```
cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for x86_64-linux-thread-multi
Work directory is /home/u1/.cpanm/work/1449501330.6888
You have make /usr/bin/make
You have LWP 6.15
You have /usr/bin/tar: tar (GNU tar) 1.26
Copyright (C) 2011 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
!
! Can't write to /usr/local/share/perl5 and /usr/local/bin: Installing modules to /home/u1/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl5 and /usr/local/bin)
!   - Configure local::lib your existing local::lib in this shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
--> Working on ./xs
Entering /home/u1/lucru/3d/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4214)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Configuring /home/u1/lucru/3d/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have Module::Build 0.3601 ... Yes (0.4214)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /home/u1/.cpanm/work/1449501330.6888/build.log for details. Retry with --force to force install it.
```

a4jp-com, 2015-12-09:
I just downloaded again but still can't compile. Any ideas on how to get this to compile? Should I try reinstalling something?

cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\design/.cpanm/work/1449636695.26636
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.13
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/slic3r-4/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4212)
Running Build.PL
Configuring C:/slic3r-4/xs ... 'cc' is not recognized as an internal or external command,
operable program or batch file.
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Module::Build 0.3601 ... Yes (0.4212)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\slic3r\GUI\GUI.o" "src\slic3r\GUI\GUI.cpp"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\libslic3r\ClipperUtils.o" "src\libslic3r\ClipperUtils.cpp"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\clipper.o" "src\clipper.cpp"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\libslic3r\LayerRegion.o" "src\libslic3r\LayerRegion.cpp"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\poly2tri\sweep\cdt.o" "src\poly2tri\sweep\cdt.cc"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\poly2tri\sweep\sweep.o" "src\poly2tri\sweep\sweep.cc"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\libslic3r\SVG.o" "src\libslic3r\SVG.cpp"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_ [... cut ...]
src\libslic3r\GUI\GUI.cpp:35:45: error: expected unqualified-id before '(' token
     SetThreadExecutionState(EXECUTION_STATE.ES_CONTINUOUS);
                                             ^
error building xs.dll file from 'src\libslic3r\GUI\GUI.cpp' at C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\Users\design\.cpanm\work\1449636695.26636\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

C:\slic3r-4>

alexrj, 2015-12-09:
What commit is that? It doesn't look like the current master.

a4jp-com, 2015-12-09:
I'm re-downloading and trying "Latest commit e2f2761" now.

a4jp-com, 2015-12-09:
It didn't work again.

C:\slic3r-4>perl Build.PL --v
App::cpanminus is up to date. (1.7039)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.05)
Encode is up to date. (2.78)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.10)
ExtUtils::ParseXS is up to date. (3.30)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.60)
Getopt::Long is up to date. (2.48)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (121)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.000002)
POSIX is up to date. (1.53)
Scalar::Util is up to date. (1.42)
Test::Harness is up to date. (3.35)
Test::More is up to date. (1.001014)
Thread::Semaphore is up to date. (2.12)
Time::HiRes is up to date. (1.9728)
Unicode::Normalize is up to date. (1.24)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (2.02)
Warning: Build.PL has been altered.  You may need to run 'perl Build.PL' again.
Cleaning up build files
Cleaning up configuration files
Not in MANIFEST: src/boost/aligned_storage.hpp
Not in MANIFEST: src/boost/array.hpp
Not in MANIFEST: src/boost/assert.hpp
Not in MANIFEST: src/boost/bind.hpp
Not in MANIFEST: src/boost/bind/arg.hpp
Not in MANIFEST: src/boost/bind/bind.hpp
Not in MANIFEST: src/boost/bind/bind_cc.hpp
Not in MANIFEST: src/boost/bind/bind_mf2_cc.hpp
Not in MANIFEST: src/boost/bind/bind_mf_cc.hpp
Not in MANIFEST: src/boost/bind/bind_template.hpp
Not in MANIFEST: src/boost/bind/mem_fn.hpp
Not in MANIFEST: src/boost/bind/mem_fn_cc.hpp
Not in MANIFEST: src/boost/bind/mem_fn_template.hpp
Not in MANIFEST: src/boost/bind/mem_fn_vw.hpp
Not in MANIFEST: src/boost/bind/placeholders.hpp
Not in MANIFEST: src/boost/bind/storage.hpp
Not in MANIFEST: src/boost/call_traits.hpp
Not in MANIFEST: src/boost/cerrno.hpp
Not in MANIFEST: src/boost/checked_delete.hpp
Not in MANIFEST: src/boost/chrono/chrono.hpp
Not in MANIFEST: src/boost/chrono/clock_string.hpp
Not in MANIFEST: src/boost/chrono/config.hpp
Not in MANIFEST: src/boost/chrono/detail/inlined/chrono.hpp
Not in MANIFEST: src/boost/chrono/detail/inlined/mac/chrono.hpp
Not in MANIFEST: src/boost/chrono/detail/inlined/posix/chrono.hpp
Not in MANIFEST: src/boost/chrono/detail/inlined/win/chrono.hpp
Not in MANIFEST: src/boost/chrono/detail/is_evenly_divisible_by.hpp
Not in MANIFEST: src/boost/chrono/detail/static_assert.hpp
Not in MANIFEST: src/boost/chrono/detail/system.hpp
Not in MANIFEST: src/boost/chrono/duration.hpp
Not in MANIFEST: src/boost/chrono/system_clocks.hpp
Not in MANIFEST: src/boost/chrono/time_point.hpp
Not in MANIFEST: src/boost/concept/assert.hpp
Not in MANIFEST: src/boost/concept/detail/backward_compatibility.hpp
Not in MANIFEST: src/boost/concept/detail/borland.hpp
Not in MANIFEST: src/boost/concept/detail/concept_def.hpp
Not in MANIFEST: src/boost/concept/detail/concept_undef.hpp
Not in MANIFEST: src/boost/concept/detail/general.hpp
Not in MANIFEST: src/boost/concept/detail/has_constraints.hpp
Not in MANIFEST: src/boost/concept/detail/msvc.hpp
[... 1683 lines cut ...]
                                                                                  ^
In file included from C:/Strawberry/c/i686-w64-mingw32/include/minwindef.h:163:0,
                 from C:/Strawberry/c/i686-w64-mingw32/include/windef.h:8,
                 from C:/Strawberry/c/i686-w64-mingw32/include/Windows.h:69,
                 from src\libslic3r\GUI\GUI.cpp:6:
src\libslic3r\GUI\GUI.cpp:25:83: error: expected unqualified-id before '(' token
     SetThreadExecutionState(EXECUTION_STATE.ES_DISPLAY_REQUIRED | EXECUTION_STATE.ES_CONTINUOUS);
                                                                                   ^
src\libslic3r\GUI\GUI.cpp: In function 'void Slic3r::GUI::enable_screensaver()':
src\libslic3r\GUI\GUI.cpp:35:44: error: expected primary-expression before '.' token
     SetThreadExecutionState(EXECUTION_STATE.ES_CONTINUOUS);
                                            ^
In file included from C:/Strawberry/c/i686-w64-mingw32/include/minwindef.h:163:0,
                 from C:/Strawberry/c/i686-w64-mingw32/include/windef.h:8,
                 from C:/Strawberry/c/i686-w64-mingw32/include/Windows.h:69,
                 from src\libslic3r\GUI\GUI.cpp:6:
src\libslic3r\GUI\GUI.cpp:35:45: error: expected unqualified-id before '(' token
     SetThreadExecutionState(EXECUTION_STATE.ES_CONTINUOUS);
                                             ^
error building xs.dll file from 'src\libslic3r\GUI\GUI.cpp' at C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\Users\design\.cpanm\work\1449685288.30180\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

C:\slic3r-4>

a4jp-com, 2015-12-09:
build.log

cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\design/.cpanm/work/1449685288.30180
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.13
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/slic3r-4/xs
Checking if you have Module::Build 0.38 ... Yes (0.4212)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Configuring C:/slic3r-4/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have Module::Build 0.3601 ... Yes (0.4212)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\Users\design\.cpanm\work\1449685288.30180\build.log for details. Retry with --force to force install it.

a4jp-com, 2015-12-10:
Should I just delete all the files in the folder instead of doing a rewrite when copying?

alexrj, 2015-12-10:
You were not compiling the latest commit.
You can just issue `git pull`.

a4jp-com, 2015-12-10:
Can you fix the download button link? I will try that though. Thank you. If it works then I'll close this thread ^^

alexrj, 2015-12-10:
What download button should I fix?

a4jp-com, 2015-12-10:
I clicked the download file button from this page. I have no idea why it isn't the newest. It should be and I've never had trouble before. Is this a github.com bug?

[image]

a4jp-com, 2015-12-10:
I just tried GitHub through the Windows program but it has the same problem.

a4jp-com, 2015-12-10:
I had to sync. Trying again now. Sorry.

a4jp-com, 2015-12-10:
[image]

Something is weird. I can't see the icons you made so I guess this is an old file too. Or is that a bug or something?

I'm not sure why I couldn't download the newest files but I will use GitHub on my desktop if that is better. I just figure downloading the ZIP files works better and that way I always have a backup if the newest version doesn't work.

simonkuehling, 2015-12-11:
Hi @alexrj,

i am having similar issues compiling the latest commit 37d05588733d0b734ea05712f65a8f95271da953 on ubuntu 14.04

My build.log is
```
cpanm (App::cpanminus) 1.7039 on perl 5.018002 built for x86_64-linux-gnu-thread-multi
Work directory is /home/simon/.cpanm/work/1449828564.720
You have make /usr/bin/make
You have LWP 6.13
You have /bin/tar: tar (GNU tar) 1.27.1
Copyright © 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL Version 3 oder später <http://gnu.org/licenses/gpl.html>
Dies ist freie Software: Sie dürfen sie ändern und weiter verbreiten.
Es gibt KEINERLEI GARANTIE, so weit das Gesetz es erlaubt.

Geschrieben von John Gilmore und Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/simon/RepRap/Slic3r/slic3r-git/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Checking if you have Module::Build 0.38 ... Yes (0.4204)
Configuring /home/simon/RepRap/Slic3r/slic3r-git/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have Module::Build 0.3601 ... Yes (0.4204)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /home/simon/.cpanm/work/1449828564.720/build.log for details. Retry with --force to force install it.
```

alexrj, 2015-12-11:
Hi @simonkuehling, I'll try to recompile on Ubuntu as soon as I can. In the mean time, what's the compilation output? It's in the mentioned build.log or in the output of `perl Build.PL --xs`

a4jp-com, 2015-12-11:
I downloaded the ZIP file after the Github update and now I can compile but I'm not sure if it is the latest version. Is there a way to check what version the zip files are? All the menu icons are missing though.

simonkuehling, 2015-12-11:
Hi @alexrj, the build.log is what i have included above (self-referencing, interestingly), but i did a <code>perl Build.PL --xs</code> which results in:

```
[...]
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -Isrc/libslic3r -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/Flow.o src/libslic3r/Flow.cpp
src/libslic3r/Flow.cpp: In member function ‘float Slic3r::Flow::spacing(const Slic3r::Flow&) const’:
src/libslic3r/Flow.cpp:57:40: error: ‘assert’ was not declared in this scope
     assert(this->height == other.height);
                                        ^
error building src/libslic3r/Flow.o from 'src/libslic3r/Flow.cpp' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /home/simon/.cpanm/work/1449835950.1871/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
```

a4jp-com, 2015-12-11:
Maybe you can try the same thing as me @simonkuehling in the meantime.

Download Github for the PC and then save the files to a folder on the C drive. Then try compiling there. My software compiles now but the icons are missing. I'll leave this thread open for now but if it gets fixed I can close this thread anytime you like @alexrj

simonkuehling, 2015-12-11:
@a4jp-com nope, same error for the zip download. So i would assume that your zip file is of an older version, maybe? You could check if there are any older zip files of the same file name in your download directory and delete them first...

a4jp-com, 2015-12-11:
I though you were the one with the compiling problem @simonkuehling .

My files compiled after the GitHub download.

simonkuehling, 2015-12-11:
@a4jp-com yes, i have compiling problems as described above - both by git sync (pull) and when manually downloading the zip snapshot.

If you cannot compile the latest git commit but compiling the zip works for you, then chances are that you may have unpacked and compiled an older version and not the latest snapshot from https://github.com/alexrj/Slic3r/archive/master.zip - especially if you miss the icons as you said. There is no reason why the zip should work, but the git sync doesn't, in my opinion...

Just for verification, could you quickly double check that? (By downloading, unpacking and compiling the zip again)... I think this would aid in troubleshooting here

a4jp-com, 2015-12-11:
I just downloaded the files and put it in a blank folder. It should be ready in a few minutes. Back soon.

a4jp-com, 2015-12-11:
[image]

That compiled but without the menu icons.

alexrj, 2015-12-11:
Lack of icons might depend on the version of wxWidgets and/or the Wx module you're using. Try to update both:

```
cpanm Alien::wxWidgets
cpanm Wx
```

a4jp-com, 2015-12-11:
It says they are up to date. Mmmm. Very strange.

[image]

a4jp-com, 2015-12-11:
How can I uninstall the Boost files and reinstall them in C:\Dev? That is the only thing I can think of that is different...

alexrj, 2015-12-11:
What's your wxWidgets version? `perl slic3r.pl --debug --gui` will tell you.

alexrj, 2015-12-11:
@simonkuehling, I just pushed a commit. Does it fix compilation for you?

simonkuehling, 2015-12-11:
@alexrj, compilation is now successful with your latest commit!

Unfortunately i do not see the icons like @a4jp-com, either...

My wxWidgets version is
```
wxWidgets version wxWidgets 2.8.12, Wx version 0,9927
```

justmyopinion, 2015-12-12:
I have  a different problem with latest (before commit making Slic3r crash with any model but do have color icons
[image]
)

alexrj, 2015-12-12:
@justmyopinion, those are not issues in Slic3r but in dependencies. You can issue `cpanm -v IO::stringy` to see why that module doesn't install successfully. Regarding XML::SAX::ExpatXS, that's an optional module which provides AMF format parsing. You probably need to install libexpat-dev.

justmyopinion, 2015-12-12:
Ok, but that did not help me in my world..
[image]

a4jp-com, 2015-12-12:
Here is the info:

C:\Slic3r>perl slic3r.pl --debug --gui
wxWidgets version wxWidgets 3.0.2, Wx version 0.9927
Data directory: C:\Users\design\AppData\Roaming\Slic3r

a4jp-com, 2015-12-12:
C:\Slic3r>cpanm -v IO::stringy
cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\design/.cpanm/work/1449912625.10308
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 2.04
Searching IO::stringy () on cpanmetadb ...
! Finding IO::stringy on cpanmetadb failed.
Searching IO::stringy () on metacpan ...
! Could not find a release matching IO::stringy () on MetaCPAN.
Searching IO::stringy on mirror http://www.cpan.org ...
Downloading index file http://www.cpan.org/modules/02packages.details.txt.gz ...
Uncompressing index file...
! Finding IO::stringy () on mirror http://www.cpan.org failed.
! Couldn't find module or a distribution IO::stringy

C:\Slic3r>

GregoryNoel, 2015-12-12:
Hi, I have the same compiling issue.
I updated with git pull 
I already tried to relaunch with --force, but nothing changed

The error is : 
-----------------
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
cc -I/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE -xc++ -Isrc/libslic3r -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -c -arch x86_64 -arch i386 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -Os -o src/libslic3r/PrintRegion.o src/libslic3r/PrintRegion.cpp
In file included from src/libslic3r/PrintRegion.cpp:1:
In file included from src/libslic3r/Print.hpp:8:
In file included from src/libslic3r/BoundingBox.hpp:5:
src/libslic3r/Point.hpp:116:10: fatal error: 'boost/polygon/polygon.hpp' file not found
#include <boost/polygon/polygon.hpp>
         ^
1 error generated.
error building src/libslic3r/PrintRegion.o from 'src/libslic3r/PrintRegion.cpp' at /System/Library/Perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /Users/Greg/.cpanm/work/1449912471.9317/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

This is the log : 
--------------------
cat /Users/Greg/.cpanm/work/1449912471.9317/build.log
cpanm (App::cpanminus) 1.7039 on perl 5.018002 built for darwin-thread-multi-2level
Work directory is /Users/Greg/.cpanm/work/1449912471.9317
You have make /usr/bin/make
You have LWP 6.13
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
--> Working on ./xs
Entering /Users/Greg/git/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4003)
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Configuring /Users/Greg/git/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have Module::Build 0.3601 ... Yes (0.4003)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have Test::More 0 ... Yes (1.001014)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /Users/Greg/.cpanm/work/1449912471.9317/build.log for details. Retry with --force to force install it.

a4jp-com, 2015-12-12:
Hi @GregoryNoel, can you see if you can compile the files in a new folder? Maybe some older files are causing a bit of a problem. 

If it does work after that please post a message here,

GregoryNoel, 2015-12-12:
Exactly the same problem with a new folder

justmyopinion, 2015-12-12:
Well i took notice from #2815 and used --force IO::Scalar and that installed IO but did not remove the error, but I had a compiled version 1.30dev of Slic3r, but it is not stable , and crashed even by opening "about" dialog only.:-(.

a4jp-com, 2015-12-14:
Sorry I missed the bit of the sentence where you said I might need to install libexpat-dev. How do I do this on a Windows machine?

I can only find simple Linux and Mac instructions.

Installing the prerequisites on Ubuntu, Debian and the likes
Installing the prerequisites on RedHat, Fedora etc.
Installing the prerequisites on Mac OS X

a4jp-com, 2015-12-16:
I have just done a recovery and reset Windows 10 (64 bit), reloaded the 32 bit of Strawberry, installed the Boost C++ libraries in the dev folder and hope everything works properly now.

a4jp-com, 2015-12-16:
The program compiles perfectly I'm just missing the icons. Can you explain how to get libexpat-dev if it will make them work?

a4jp-com, 2015-12-16:
I'm running Windows 10 (64-bit).

lordofhyphens, 2016-03-13:
http://expat.sourceforge.net/ is the website for expat, and their source for win32 is there. I don't know myself if expat is used in the icon loader (although it might).

alexrj, 2016-03-13:
expat is only used for speeding up the loading of AMF files, but if I recall correctly it's not used on Windows at the moment.

closed 2016-03-26 by alexrj