slic3r/Slic3r #3149: Arrange button not working

koghe, 2015-12-06:
Hi
load 2 different objects, press "Arrange" , nothing happen.
current git
linux ubuntu 15.04

ps.while triyng i cannot select 2 objects from the list panel.

lordofhyphens, 2015-12-06:
One of the reasons that can happen is that the logic can't find a place to
put everything.

koghe, 2015-12-06:
It work for you? even for simple object? with 2 cubes doesn't work, also with "More" option and duplicated objects

kefir-, 2015-12-06:
Slic3r arranges as you load, so there's nothing to change when you click arrange if you haven't done anything else. Have you moved the objects first, and then tried clicking arrange? Does something happen then?

koghe, 2015-12-06:
No, just load 2 object one after other and press the button nothing happen, or try with this object 
load, it then right click and select split, click on arrange and nothing...
https://www.dropbox.com/s/3q298h7d7jujdik/coolingcase.stl?dl=0

i see alex is working on a new gui, will wait for it :)

lordofhyphens, 2015-12-06:
I actually disable auto arrange myself. It can be a headache.

koghe, 2015-12-06:
:) yes but sometmes is useful to arrange multiple objects, np will wait

koghe, 2015-12-08:
with last commits it work better 
open
[image]

split

[image]

Arrange first click 

[image]

Click Arrange again.. ok! 

[image]

koghe, 2015-12-15:
after this fix arrange work with one click only https://github.com/alexrj/Slic3r/commit/c73378744f0d6e8c4a07952d9228069f35fcb56e

manually arranged
[image]

click to Arrange

[image]

it work better now

alexrj, 2015-12-19:
Ok.

closed 2015-12-19 by alexrj