slic3r/Slic3r #3150: [Proposal] Save Configuration Slic3r on gcode

Nitrosito, 2015-12-06:
Hello there! The first congratulations and thanks for the program

It would be nice to write configuration on  .gcode file is printed, and can import settings from the .gcode.

I think it easy, would simply export the configuration as a comment at the beginning or end of .gcode, and to import reading if there is a "magic string" in the .gcode

xoan, 2015-12-06:
Indeed, Slic3r already does that (please, take a look at the end of any .gcode file)

Nitrosito, 2015-12-06:
But, it possible import this configuration to Slic3r program?

simonkuehling, 2015-12-06:
@Nitrosito, this is actually a very neat idea! As all parameters are already added as a comment at the end of a gcode file today, that's probably quite easy to implement.

It would eliminate the need to manually store the configuration along with the resulting gcode for users that do 3D print service and the like - and need to recover job-specific parameters from earlier print jobs at a later date.
Also, debugging and bug reports get a little more convenient as you would only need the STL+gcode without exportint a separate config then.

xoan, 2015-12-06:
So, for better understanding, I guess the issue title should be changed for some more descriptive statement :)

I think it would be a cool feature too.

Nitrosito, 2015-12-06:
I think I could easily generate a bash script to get the information, generate a file, and import it.

But would remain much more elegant dialogue in the program to import.

As with the printing time. I see what it will take to print use this script in post-scripts [on my repo](https://github.com/Nitrosito/3D/tree/master/ScriptTiempoSlic3r) is based on GCODER, it would be very interesting to implement in serial to show How soon somewhere

alexrj, 2015-12-06:
See #2010

kefir-, 2015-12-06:
It'd also be nice if the stl filename(s) were stored in the comments, so that they could be automatically reloaded. Ok, and now I'm maybe getting carried away, but rotation, scaling and placement of all the parts would also be really nice. Then it'd be easy to make minor modifications to large prints, or reslicing a full bed after redesigning a single part. 

If the new parts collide, the user will have to resolve that manually.

Paul--W, 2015-12-08:
Alex,
I think that allowing settings to be imported from the gcode is very important for debugging purposes as well as the other purposes mentioned.  How many times have you had to ask for config files...

There is a solution to the issue you raised in #2010, "The problem is just about the GUI: there's only a single "Filament preset" tab, so we can show a single preset at time. There's just no friendly way to show more than one "stored but not saved" preset."

When importing the settings from the gcode you could compare them to current settings in the existing presets.  If the settings are not the same as the presets, import them with a name like "Import<datetime>" and save them (not dirty). Then just allow presets to be renamed.  The presets used should be set as the active/current presets.

You could save the names of the Print Setting, Filament Setting, and Printer setting presets in the gcode also.  Then if the preset does not exist the previous name could be used when importing settings.  Group the individual settings under the preset name in the gcode comments.

Nitrosito, 2015-12-10:
So in the end it is going to add this feature functionality Slic3r?