slic3r/Slic3r #3151: Print setting selector box too small

thecrazy, 2015-12-07:
I like to use long descriptions for my print settings, could you please allow us to resize the column and have the selection grow and shkring with it?

See this pic (I would love to be able to grap the separator in just drag it around like we often see in other apps
[image]

alexrj, 2015-12-18:
I implemented this, but wxWidgets does not provide a nice way for it as it doesn't resize live while updating. However it works for now.

thecrazy, 2015-12-18:
Awesome. Thank you very much.

xoan, 2015-12-19:
I think it is only a Windows related behaviour.

Using Slic3r 1.2.9 with Wx 3.0.2 and libwx-perl 0.9923 we can see this:

[image]

As expected, the length of the dropdown menu is not related to the length of the dropdown button.

thecrazy, 2015-12-19:
Oh, that would work fine too.

alexrj, 2015-12-19:
Yes, this is a Windows specific issue.

farhaven, 2015-12-23:
The pull request seems to break the column display on OpenBSD with p5-Wx-0.9927. The content of the column is always zero-width so it can't be interacted with. Moving the separator does resize and move the right column, but the content of the left column stays zero-width.

ednisley, 2015-12-24:
The contents remain hidden in Xubuntu 14.04 after dragging the column width to the right:

[image]

Compiled from [master](7c8b71012) on 22 December.

alexrj, 2015-12-25:
I reverted this.

Love to hear someone is running Slic3r on OpenBSD.