slic3r/Slic3r #3155: git master compile failure

wolfmanjm, 2015-12-11:
```
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -Isrc/libslic3r -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/Flow.o src/libslic3r/Flow.cpp
src/libslic3r/Flow.cpp: In member function ‘float Slic3r::Flow::spacing(const Slic3r::Flow&) const’:
src/libslic3r/Flow.cpp:57:40: error: ‘assert’ was not declared in this scope
     assert(this->height == other.height);
                                        ^
error building src/libslic3r/Flow.o from 'src/libslic3r/Flow.cpp' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
```

On ubuntu 14.04 32bit, perl v5.18.2, gcc version 4.8.4

alexrj, 2015-12-11:
Thank you! Does my latest commit fix this?

wolfmanjm, 2015-12-11:
Yes, works great now... thank you ;)

closed 2015-12-11 by wolfmanjm